carla-simulator/carla #4012: Client-side white screen flickering in low quality mode

wuhanstudio, 2021-03-23:
Hi, 

I started a low quality Carla server using:

```
 ./CarlaUE4.sh -quality-level=Low
```

Then I connect to a Carla Server (Ubuntu 18.04) from a Windows Client using:

```
python manual manual_control.py --host xxx.xxx.xxx.xxx
```

**Server Side**: Ubuntu 18.04, GeForce RTX 2080Ti, CARLA_0.9.11
**Client Side**: Windows 10, Intel Graphics, CARLA_0.9.11, Python API

https://user-images.githubusercontent.com/15157070/112177308-b29d5780-8bf0-11eb-8ac5-cfb407991b58.mp4

The RGB Camera flickers as shown in the video, while the depth camera and other sensors work fine.

The same issue arises when I connect locally (127.0.0.1) as long as the Carla server starts in low quality mode.

Is the issue related to Carla, Python API or the Unreal engine?

Thanks for your help.
Han

corkyw10, 2021-03-30:
@bernatx could you have a look at this please?

XlightwolfX, 2021-04-20:
Hi, I was wondering whether there are any new findings regarding this issue. I am experiencing exactly the same.
CARLA: 0.9.11
Ubuntu 20.04
GeForce GTX 960M
Both server and client running on one machine

Note that in my case I experienced this only when running CARLA from the deb installation. When I build CARLA from source and start the server via UE4 then the glitching of the RGB camera disappears. The problem reappears, however, if I package the project (`make package` or UE4 GUI, makes no difference) and start CARLA through the created binary.

bogdan-kovalchuk, 2021-04-22:
It seems that this issue has been added in recent versions, at least I've checked versions 0.9.10 and 0.9.11 and it is already there. In version 0.9.6 everything works fine and there is no flickering.

hynkis, 2021-05-02:
In my case, when I started the simulator using './CarlaUE4.sh -opengl -quality-level=Low', there is always the flickering issue.
However, using './CarlaUE4.sh -quality-level=Low' without 'opengl' option, there is no flickering issue.

JRestovich, 2021-09-15:
Changing -opengl for -vulkan the problem was solved at least in my case:

./CarlaUE4.sh -vulkan -quality-level=Low

closed 2023-06-02 by stale[bot]