carla-simulator/carla #4015: Camera still receives data in no-rendering mode in 0.9.11

Chiakiaki, 2021-03-24:
When enabling no-rendering mode in 0.9.11, the server side turns to black screen as expected. But on client side, I can still see image returned by camera sensors, and the FPS does not improve. Is this normal?

I'm an old user. I previously use version 0.9.8, and recently update to 0.9.11.  In same situations with same code, camera in 0.9.8 will return black screen or mosaic.

corkyw10, 2021-03-30:
@DSantosO could you follow up on this please?

DSantosO, 2021-04-01:
Hello @Chiakiaki,
We will take a look at that. In the meantime, if you don't need any cameras you can also try the option '-nullrhi'. This should remove totally the render thread and provide better performance.

Chiakiaki, 2021-04-20:
@DSantosO 
Thank you. 
I tried your advice, and found that the key of improve fps is turning off camera sensor in no-rendering mode (30fps -> 38fps for me). With above setting, '-nullrhi' doesn't bring more significant improvement, maybe there are already no rendering computation there.

closed 2021-04-20 by Chiakiaki