carla-simulator/carla #4018: How to load a new world without having diverging views?

cinjon, 2021-03-24:
How do I load a new world while Carla is running? I have tried a lot and nothing has worked, e.g the following, which results in a world that diverges from what is visible on the screen. It was the cause of #4013. 

I need this because my model decides what map to use and so I need to be able to change the world on the fly.

```
def foo(client, world, ...):
    <Do stuff>
    world = client.load_world(new_map_name)
    world.tick()
    <Do more stuff>
    <Return this new world along with the other results and continue>
```

corkyw10, 2021-03-30:
@Axel could you follow up on this please?

Axel1092, 2021-04-13:
Hi @cinjon, I don't understand your problem. The function `world = client.load_world(new_map_name)` should do exactly what you need. The `new_map_name` needs to be one of the existing maps such as `Town01`, `Town02`, etc. Notice that we also offer `Town01_opt` versions of the maps which offer a small performance improvement.

closed 2023-06-02 by stale[bot]