carla-simulator/carla #4019: record and save .pcd file as a stream

cacent96, 2021-03-25:
I try to look up on the website and most of the file type for the lidar are the .ply files. Would it possible to save it as a point cloud data format(pcd)? If yes, is that anyway I could save the file as a stream but not in image? Beside, how could I open the .pcd file?

jarvis-huang, 2021-03-26:
I don't know what you mean by a stream. Normally a pcd is simply a single pointcloud capture at one time instant.
You can save .pcd using pypcd. You need to write a small script to do the conversion.
You can view .pcd in rviz (again need a small script) or CloudCompare (directly open).

cacent96, 2021-03-26:
> I don't know what you mean by a stream. Normally a pcd is simply a single pointcloud capture at one time instant.
> You can save .pcd using pypcd. You need to write a small script to do the conversion.
> You can view .pcd in rviz (again need a small script) or CloudCompare (directly open).

Hey there thank you for your reply. Since the pcd only capture at one time instant, then would it be possible to combine or merge the .pcd and turning it into a single file with multiple datasets?

closed 2021-05-10 by cacent96