carla-simulator/carla #4020: How to ego-vehicle invisible?

foamliu, 2021-03-25:
The front of the car takes up too much space.

You need:
[image]

You get:
[image]

jackbart94, 2021-03-25:
You could move the camera upwards a bit and the result would be the same as in the picture above. Do you need help with that? The issue isn't very clear.

foamliu, 2021-03-27:
Thanks @jackbart94 ，I moved the camera to new position (2.5, 0,1), in front of the car, now it is better. I was worried about the impact but seems it is under control.

closed 2021-03-27 by foamliu