carla-simulator/carla #4022: how to use setters and getters in carla

parth3153, 2021-03-25:
i know this is not an issue but i am not getting any answers in the discord community and the documentation is not enough..any developers can you provide me with an example script.. i have been stuck on thins since 2 weeks now...any help appreciated..thank you

ar4development, 2021-03-25:
@parth3153 can you add more context to your question? are you asking about Python API or about contributing to Carla codebase? Can you describe your case you stuck with?

ParthAD3, 2021-03-29:
hello,
so i want to get the velocity, acceleration, and torque curve plots of my simulation log files i dont know how to extract data from my log files.

so i wrote this script.. to print the velocity and at the end plot these data at the end, the isuuee is i am not getting any plots, the vehicle is not moving even tho is was set to autopilot, so i just get the constant location printed, i think the script is compleately worng, my implementation.. so i am asking for an example script..

i mean yes they have given the commands in the documentation and for some thats enough but many people i have asked dont know and the one who knows are not answering..

i hope you get the problem now.  below is my script:

import carla
import os
import random

def main():
    vehicle_list = []
    try:
        client = carla.Client('127.0.0.1', 2000)
        client.set_timeout(30.0)

        world = client.load_world('Town01')
        settings = world.get_settings()
        settings.fixed_delta_seconds = 0.1
        settings.synchronous_mode = True     
        world.apply_settings(settings)
        blueprint_library = world.get_blueprint_library()

        traffic_manager = client.get_trafficmanager(8000)
        traffic_manager.set_hybrid_physics_mode(True)

        # Create ego vehicle
        bp = random.choice(blueprint_library.filter("*twizy3*"))

        while True:
            transform = random.choice(world.get_map().get_spawn_points())
            vehicle = world.try_spawn_actor(bp, transform)
            if not vehicle is None:
                break

        vehicle_list.append(vehicle)
        vehicle.set_autopilot(True, traffic_manager.get_port())

        for _ in range(200):
            world.tick()

            location = vehicle.get_location()
            velocity = vehicle.get_velocity()
            control = vehicle.get_control()

            print('location', location.x, location.y, location.z)
            print('velocity', velocity.x, velocity.y, velocity.z)
            print('control', control.throttle, control.brake, control.steer)

    # Set autopilot to False, destroy all actors and quit
    finally:
        for actor in vehicle_list:
            actor.destroy()

if __name__ == '__main__':
    main()

i wait for your answer, 
thankyou @ar4development

ParthAD3, 2021-03-29:
also, 

i found this script here https://github.com/carla-simulator/carla/issues/3256

but i am getting index error here at line 87 and 104 

[image]

corkyw10, 2021-03-30:
Hello @ParthAD3 my assumption is that are getting zero values for velocity because you are using `traffic_manager.set_hybrid_physics_mode(True)` which means vehicles farther than a certain radius from the ego vehicle will have their physics disabled, so their velocity would be zero. it could also be related to this issue #3683

closed 2021-11-18 by jackbart94