carla-simulator/carla #4024: Importing very simple and clean XODR makes Carla crash

ar4development, 2021-03-25:
I have prepared quite a simple map which was migrated from OSM using `netconvert` and then added some manual changes to the resulting file. This map is attached to the issue.

When I try to load this map into Carla using `python config.py -x /path/to/map` the simulator is getting crashed. I looked through some sort of similar issues here and they seem to be related to breaches between the lanes. However I carefully examined my map in QGIS and took various routing tests there and all seems to be fine from routing point of view.

One more thing is that experimentally I figured out that the crash is caused by this part of my map:

[image]

Since there is no logs provided by Carla I cannot give more insight on the issue. I'm using Carla 0.9.11 with RSS support.

Appreciate your help.
[map.zip](https://github.com/carla-simulator/carla/files/6205523/map.zip)

corkyw10, 2021-03-30:
@Axel1092 could you follow up on this please?

Axel1092, 2021-04-13:
Hi @ar4development, OpenDRIVE maps generated by netconvert tend to be broken due to errors in the junction record. We are developing our own conversion solution to provide OSM support in CARLA (https://carla.readthedocs.io/en/0.9.11/tuto_G_openstreetmap/) hopefully this will help.

ar4development, 2021-04-14:
Hi @Axel1092 , what do you mean by "errors"? Doesn't passing routing test in QGIS mean there is no junction error in this particular attached map?

Axel1092, 2021-06-30:
Hi @ar4development, sorry for the superlate reply. We have seen that netconvert has bugs regarding road connections. The resulting OpenDRIVE has lanes connecting non-existing lanes which causes the crash in CARLA.

ar4development, 2021-07-19:
@Axel1092 better later than never :) thanks for your replies. Is there a way I could figure out which section in my particular file is problematic? Put an extra log to the code or a break point? How do I debug that?

closed 2022-04-29 by stale[bot]