carla-simulator/carla #4026: Non-determinism with Carla in synchronous mode

Stephanehk, 2021-03-25:
I am running two trials of a Vehicle (spawned at the same spawn point with the same weather conditions) with the action (throttle = 0.5, steer = 0, brake = 0) at a frame rate of 10hz for 30 timesteps. The vehicle is simply going straight down a road. When using vehicle.get_transform() and vehicle.get_velocity(), however, both trials lead to different vehicle positions and different velocities at each timestep. I am running Carla 9.11 in synchronous mode. 

30th timestep of trial 1:
'agent_x': [88.6199722290039], 'agent_y': [278.54150390625], 'agent_z': [0.0015868949703872204]
'velocity_x': [5.553758e-06], 'velocity_y': [3.372087]

30th timestep of trial 2:
'agent_x': [88.6199722290039], 'agent_y': [278.20947265625], 'agent_z': [0.0015838050749152899]
'velocity_x': [5.5240553e-06], 'velocity_y': [3.2560413]

jackbart94, 2021-03-26:
First, you should work with a fixed_time_step of 0.05, due to the physics engine and the Traffic Manager. This will probably fix your issue.

If this doesn't work, please explain more in detail how you are activate synchronous mode and determinism within Traffic Manager.

Stephanehk, 2021-03-26:
Setting the fixed_time_step to 0.05 solved the issue.

closed 2021-03-26 by Stephanehk