carla-simulator/carla #4027: Running server and client in different machines has very low fps

zjh6252007, 2021-03-26:
I am trying to running server and client in two computers. One running server and another connect to that server. I found the fps is very low about 20-30. When I running server and client in same computer I got about 80fps. So does anyone know what caused the decrease in FPS？

thytless, 2021-03-26:
How could we reproduce your problem? More information is appreciated, including but not limited to:

- Did you use a script in carla repository or you have your own? 
- Are you in synchronous mode or asyn?
- And what about your machine environment(both server and client)? What about the bandwidth?

zjh6252007, 2021-03-26:
@thytless Hello I didn't use any script. I just connected with manual_control.py.
I think I used the asyn cause I disabled synchronous with config.py. 
I'm working on windows and working with Carla package and I connected the two computers using the network cable.
Thank you.

qhaas, 2021-04-02:
The machine running carla client have an inferior CPU compared to the one running carla server?  Significant CPU processing still occurs client side depending on what you are doing.

DSantosO, 2021-04-26:
Hello @zjh6252007,
I assume that you are talking about the server FPS that the `manual_control` is showing, right?
If you are in async mode (as the manual control is by default), the fps are given by the power of the machine so I would ask the same as @qhaas, how are the machines in which you are trying to run the server in terms of performance?

If you like you can benchmark both machines without the manual_control with our [benchmark script](https://carla.readthedocs.io/en/latest/adv_benchmarking/) to see how they compare.

zjh6252007, 2021-05-03:
Hello @DSantosO 
I used the manual_control. The configuration of the server and client is exactly the same. When I ran the client, the fps shows on the screen is about 70-80 but I do feel it can only be 20-30 fps. 

I also found that the network rate seemed to have a big impact on the fps. I first used WiFi to connect the server. Even though the number of frames on the screen is 70-80, the actual fps is less than 5 fps. After I connected the two computers with cable, the fps increased to 20-30.

aadishdnaik, 2021-11-15:
@zjh6252007 can you tell me how to run the server and client on different machines? I want to run the server on Windows and the client on Ubuntu. Even if you could point me to a resource online/documentation on how to do it, that would be great! 

> I am trying to running server and client in two computers. One running server and another connect to that server. I found the fps is very low about 20-30. When I running server and client in same computer I got about 80fps. So does anyone know what caused the decrease in FPS？

closed 2022-04-29 by stale[bot]

ManishwarG, 2023-03-07:
> @zjh6252007 can you tell me how to run the server and client on different machines? I want to run the server on Windows and the client on Ubuntu. Even if you could point me to a resource [...]

Hey @aadishdnaik  even I'm trying to run both on different machines have you found any documentation in this regard. Any help would be appreciated.