carla-simulator/carla #4029: unit of distance

abirbensoltane, 2021-03-26:
Hello! I want to know the unit of distance for example when I use carla.Location(x=0,y=7, z=2.8) I want to know the unit of z=2.8.
In addition, I'm trying to take photos of different angles and positions of a car to generate a dataset. But I don't know how to change the direction of the camera so that it can always be oriented towards the car. In blender, I use .to_track() to make the camera rotate towards the car.

corkyw10, 2021-03-30:
Hi @abirbensoltane the units of measurement are meters and I think you will find a nice example of the camera behaviour in the example script `manual_control.py` in the function `CameraManager`

abirbensoltane, 2021-03-31:
thnx @corkyw10

closed 2021-05-21 by corkyw10