carla-simulator/carla #4030: How to run faster when importing fbx(by Roadrunner) to CARLA

EchoYJJ, 2021-03-27:
Hi! I use Roadrunner to generate the detailed 3D secne (fbx file).
[image]

However, when I export it to CARLA, about 6600 files needs to be imported(mainly road barriers for my scene).
[image]
I ran for nine hours and didn't finish it.
If you know how to run faster(even sacrifice some accuracy), please give me some advices, thanks in advance!

EchoYJJ, 2021-03-27:
I've tried to simplify my scene, and increase the scale to 550. However, the importing process seems to be stuck in certain point.
[image]
It took just a little time to 258/551, but stayed unchanged for a long time.

corkyw10, 2021-03-30:
hi @EchoYJJ have a read of this [post](https://forum.carla.org/t/importing-maps-from-roadrunner-to-carla-on-linux/839/8) in the forum to see if it helps to explain more about the map import process and how to handle the limitations plus some future plans. Let us know if you have any follow up questions.

EchoYJJ, 2021-03-31:
Hi @corkyw10, thanks a lot for your advice. It seems that the map import process of complex scene still remains to be a time consuming task. Maybe in UE4.25 or UE4.26, the situation could be better.

I also trird to generate much vegetation in Roadrunner, then over 10000 files need to be imported, which is very time consuming and meaningless. As an alternative, I simplify my Roadrunner scene to only roads and grass, while spawn the vegetation and props in CARLA through **Foliage** which is suggested by XGodina in issue 3981.
https://github.com/carla-simulator/carla/issues/3981
Apart from the problem of  "how to paint the trees in stright line" in 3981, the most confusing issue to me is how to control the speed of co-simulation between SUMO and CARLA.

If my scene is simple, the simulation time equals to real world time. However, when my scene gets more complex, the simulation speed slows down(1s in real life≈5s in simulation). I also notice that vegetation in CARLA are dynamic which is unnecessary for me.  Whether the simulation speed will rise, if I change the mode from dynamic to static. 
Also, if you know other methods to speed up the simulation, please give us your advice.
Many thanks!

closed 2023-06-02 by stale[bot]