carla-simulator/carla #4031: Different on total point counts

lapd-viosoft, 2021-03-27:
Hi,
I'm using C++ example with Carla 0.9.11 to get total points from lidar and I found that, total point count are not the same on the way we count it.
```c
    // Register a callback to receive data
    lidar->Listen([](auto data) {
      auto lidar_data = boost::static_pointer_cast<csd::LidarMeasurement>(data);
      assert(lidar_data != nullptr);

      /////////////////////////////////////////////////////////////////////////
      ///////////////////////// FIRST METHOD //////////////////////////////////
      /////////////////////////////////////////////////////////////////////////
      int totalPoints1 = 0;
      for(unsigned int j = 0; j < lidar_data->GetChannelCount() ; j++){
        totalPoints1 += lidar_data->GetPointCount(j);
      }
      std::cout << "totalPoints1=" << totalPoints1  << std::endl;
      /////////////////////////////////////////////////////////////////////////
      ///////////////////////// SECOND METHOD //////////////////////////////////
      /////////////////////////////////////////////////////////////////////////    
      int totalPoints2 = 0;
      auto iter = lidar_data->begin();
      while (iter != lidar_data->end()){
          totalPoints2++;
          iter++;
      }
      std::cout << "totalPoints2=" << totalPoints2  << std::endl;

      // Just for debug
      exit(0);
    });
````

```sh
Client API version : 0.9.11
Server API version : 0.9.11
totalPoints1=12232
totalPoints2=12221
```

Can you please tell me, which one is correct way to get total points?
Thank you!

corkyw10, 2021-04-06:
@DSantosO could you follow up on this please?

DSantosO, 2021-04-07:
Hello lapd-viosoft,
Thank you for your report.
We have reproduced the issue and we are looking into it.

DSantosO, 2021-04-09:
Hello @lapd-viosoft,
There was a bug in the computation of the channel point count. It is fixed now in PR #4075 and it will be available in `dev` branch soon.
Thank you for your report.

PR #4075 (2021-04-14): Fix bug in Lidar channel point count
#### Description

This PR fixed a bug in the point count of each channel in the case that there was a dropoff based on Intensity.
The points that were being dropped based on intensity were not being removed from the channel count so the real points being sent were less than the given by `get_point_count()` method.

#### Where has this been tested?

  * **Platform(s):** Ubuntu 18.04
  * **Python version(s):** Python 3.6
  * **Unreal Engine version(s):** 4.6

Fixes #4031

#### Possible Drawbacks

None

---
This change is [[image]](https://reviewable.io/review [... cut]
closed 2021-04-26 by DSantosO