carla-simulator/carla #4032: Feature request: Simulate point cloud skew from the LiDAR when in motion

heethesh, 2021-03-27:
When the LiDAR is under motion, a real-world point cloud would have a skew or a rolling-shutter effect in the point cloud (behavior would vary from static and dynamic objects) due to the small delay (in the order of nanoseconds) between the light rays casted and received. Looks like the current LiDAR plugin freezes the scene and then performs ray-casting. Is there a way this skew could be simulated? This is similar to the Compensated flag in LGSVL simulator ([see](https://www.svlsimulator.com/docs/simulation-content/sensors-list/#lidar)).

corkyw10, 2021-04-06:
Hi @heethesh thank you for the input. @DSantosO could you have a look at this please?

DSantosO, 2021-04-26:
Hello @heethesh,
Thank you for your comment. This feature can be very interesting to add. Although it is not currently in our roadmap, we will consider it. 
I am not sure how the LGSVL simulator does it but you are right about how it works in CARLA. One possible implementation of this can be modifying the origin of the raycast for each ray. As we are doing several rays of each channel for the whole circular section that we need to cover in the last dt, we can actually modify the starting point based on the movement of the car during this last timestep.

heethesh, 2021-04-27:
@DSantosO Shifting the ray cast origin based on the vehicle motion model is a great idea! The implementation would also be computationally efficient avoiding simulation at very small time steps. However, the ray-cast returns and skew observed on dynamic objects in the environment would be incorrect in this case. I wonder if UE4 has a way to step through the motion model of only some objects such as vehicles at really small time-steps to obtain better ray cast results with the dynamic objects.

DSantosO, 2021-04-27:
Hello @heethesh,
That's true, the dynamic objects will be frozen in the current position. 
There is also another possibility. Right now, between two steps we perform a configurable number of [substeps for physical computations](https://carla.readthedocs.io/en/latest/adv_synchrony_timestep/#physics-substepping). 
There is a way of adding logic in these substeps but I don't know if we can do raycasting there.

closed 2023-06-02 by stale[bot]