carla-simulator/carla #4036: Sensor measurements not correct

zhaoshan2, 2021-03-29:
Hello,
I am trying to export the position of the waypoints of a trajectory to a .txt file then read these waypoints and follow them, so that we can have the same route every time. Meanwhile the sensor measurements(gnss, imu, lidar) and the vehicle status are recorded. But now there are two problems
1. First x,y,z of the waypoints are exported to a txt file, and then they were read and transformed to the waypoint class by
`waypoints.append(map.get_waypoint(carla.Location(x,y,z),project_to_road = False))`
And to follow these waypoints:
`waypoint = waypoints[curr_waypoint]`
`vehicle.set_trasnform(waypoint.transform)`
`curr_waypoint += 1`
But there are still shifts between the waypoints.tranform.location and the vehicle.tranform.location, is there anything we can do to calibrate the difference?

2. In the waypoint nagivation, the velocity (returned by `vehicle.get_velocity(`)) seems not correct. 
This is the plotting of the velocity
[image]

Could you help figure what the problem is?
BR,

corkyw10, 2021-04-06:
Hi @zhaoshan2 thank you for the question. 

@jackbart94 are you able to have a look at this please?

jackbart94, 2021-04-12:
Please, could you provide us the script you're using (or a similar one for the vehicle recording and plotting) and which version of CARLA you are currently working with?

zhaoshan2, 2021-04-12:
> Please, could you provide us the script you're using (or a similar one for the vehicle recording and plotting) and which version of CARLA you are currently working with?
Carla Version: Release carla 0.9.11
This script is developed from the tutorial.py. I have simplified the function and it is able to record the vehicle status such as position, velocity and acceleration and stored in the txt file. The script can be found here
https://github.com/zhaoshan2/Carla_RR/blob/794049b734cb29e09dc57cf1b0fbff5090cadfa8/PythonAPI/examples/RECORD.py
To run the script, first you need to create a folder named '_out', the recorded file will be stored in this folder. From the results you could notice that the velocity in x and y direction are all 0 (except for the first few rows)
Thanks for your time and look forward to your reply!

jackbart94, 2021-04-13:
From what I have been able to see by running and analyzing your script, I can tell that if you are using the vehicle navigation controlled by waypoints, you are not actually applying any control to the vehicle in terms of throttle, brake or steering. You are just teleporting it to the next waypoint, and this is why you can't see any velocity. Instead, if you activate the Traffic Manager, there's your velocity. 

My suggestion would be to use any of the agents in the ``automatic_control.py`` scripts and try to feed your waypoints to them, so you actually have something that moves according to your trajectory.

Let me know if I can help with anything else!

zhaoshan2, 2021-04-15:
Thanks a lot for the interpretation!

closed 2021-04-15 by jackbart94