carla-simulator/carla #4037: "recipe for target 'package' failed" Error on "make package"

PriestOfAdanos, 2021-03-29:
My log file:
[log.txt](https://github.com/carla-simulator/carla/files/6220954/log.txt)

Plz help :(

PriestOfAdanos, 2021-03-29:
replication of #3879

corkyw10, 2021-04-06:
@PriestOfAdanos thank you for the question.

@bernatx could you have a look at this please?

NielsPichon, 2021-06-03:
It looks like I am having very similar issues when trying to build from the 0.9.10 branch, where the make package recipe fails because of some missing assets (`TrafficLightPieceStructure.uasset` as well as `ST_TrafficLightHead.uasset`, like with PriestOfAdanos, but also some materials). I have tried re-running the Update.sh script but it did not help. 

Is there anywhere else than the AWS tar where I could manually retrieve these missing assets, which I think would solve this issue?

NielsPichon, 2021-06-03:
Looking more closely, it seems that the mentioned assets are there, but instead of being correctly linked, the faulty structures point to a hardcoded path `/Game/...`
There is also a number of occurences of this hardcoded path in `carla/Unreal/CarlaUE4/Config/DefaultGame.ini` and `Saved/Cooked/LinuxNoEditor/CarlaUE4/Metadata/CookedIniVersion.txt`
I think these are autogenerated? Either way they are prone to causing issues.

I'll try switching the assets and the paths to the correct one and keep you updated about whether this helps with the package recipe

NielsPichon, 2021-06-03:
So here is a complete list of all I changed, which provided your issue is the same as what I had, should allow you to package:
- In `carla/Unreal/CarlaUE4/Config/DefaultGame.ini` I switched all the `/Game/` for `<path-to-carla-repo>/Unreal/CarlaUE4/Content/Carla/`, except for the materials at the top which are in `Content/Static` rather than `Content/Carla/Static`
- In  `Saved/Cooked/LinuxNoEditor/CarlaUE4/Metadata/CookedIniVersion.txt` I swapped again the `/Game`
- Same thing in The `Content/Carla/Config` folder where I swapped `/Game` in `Town06_Opt.package.json`,  `Default.package.json`, `Town10HD_Opt.package.json`, `Town7_Opt.package.json`
- In the editor, I replaced the assets in `Content/Carla/Static/TrafficLight/StreetLights_01/TrafficLightStructure.uasset`, `Content/Carla/Static/TrafficLight/StreetLights_)1/TrafficLightPieceStructure.uasset` and `Content/Carla/Blueprints/TrafficLight/ST_TrafficLightHead.uasset` by the proper assets (the references got removed)
- Finally, I kept having ` Error: UStructProperty::Serialize Loading: Property 'StructProperty /Game/Carla/Static/TrafficLight/Streetlights_01/BP_TrafficLightNew.BP_TrafficLightNew_C:UserConstructionScript.CallFunc_Array_Get_Item_1'. Unknown structure.` which I solved by creating another Heads2  array of ST_TrafficLightStructure and replacing the first one which I then deleted. This was to force UE to update the reference to the asset and that resolved the issue.

Doing all this removed all the not found/ reference errors. Now I can't seem to be able to spot any remaining error but it still fails.
Here are the [Log.txt](https://github.com/carla-simulator/carla/files/6591272/Log.txt)  in case it helps. 

If you feel the remaining issue is unrelated to @PriestOfAdanos 's original issue, I'm happy to open a new issue.

Thanks for the help,m
Niels

hashim19, 2022-07-15:
@PriestOfAdanos Hi, were you able to export your vehicle in blender? If yes, can you explain how you achieved that?

closed 2026-11-13 by carla-triage[bot]