carla-simulator/carla #4038: Collision sensors don't work sometimes

imoneoi, 2021-03-30:
In synchronous mode, I spawned multiple cars with different blueprint, and attach a collision sensor on each of them. Some of the collision sensors work and some does not work at all.

It also appears randomly, spawning 5 different cars will reproduce this issue with high probability.

It seems that someone had the same issue, but not resolved yet:
https://github.com/carla-simulator/carla/issues/1553#issuecomment-652769315

Thanks.

Version: CARLA 0.9.11 on Ubuntu 20.04

imoneoi, 2021-03-30:
It seems that only these 3 models don't work
```
vehicle.chargercop2020.chargercop2020
vehicle.charger2020.charger2020
vehicle.mercedesccc.mercedesccc
```

Code to reproduce:
(Spawn many cars and run straight at full speed, check if collision detected)

```
import carla
import numpy as np
import tqdm

COLLISION_STATE = None

def _collision_callback(car_id, event):
    global COLLISION_STATE

    COLLISION_STATE[car_id] = True

def test_collision_sensor(
    num_cars: int = 10,
    num_test_rounds: int = 100,
    
    max_timestep: int = 5000,
    dt: float = 0.1
):
    global COLLISION_STATE

    # Create client
    client = carla.Client('localhost', 2000)
    client.set_timeout(10.0) # seconds

    for r in range(num_test_rounds):
        print("Test round {}".format(r))

        # Clear collision state
        COLLISION_STATE = np.zeros(num_cars, dtype=np.bool)

        # Load world
        world = client.load_world('Town04')

        # Synchronous tick mode
        settings = world.get_settings()
        settings.synchronous_mode = True
        settings.fixed_delta_seconds = dt
        world.apply_settings(settings)

        # Create cars with sensors
        car_list = []
        sensor_list = []
        car_name_list = []

        car_blueprints = world.get_blueprint_library().filter("vehicle.*")
        car_spawn_points = world.get_map().get_spawn_points()
        for car_id in range(num_cars):
            # get car blueprint
            car_bp = np.random.choice(car_blueprints)
    
            # get spawn point
            car_pos = np.random.choice(car_spawn_points)
            car_spawn_points.remove(car_pos)

            # spawn car
            car = world.spawn_actor(car_bp, car_pos)
            car_list.append(car)
            car_name_list.append(car_bp.id)

            # create collision sensor
            sensor_blueprint = world.get_blueprint_library().find("sensor.other.collision")
            sensor = world.spawn_actor(sensor_blueprint, carla.Transform(), attach_to=car)
            sensor.listen(lambda event, id=car_id: _collision_callback(id, event))
            sensor_list.append(sensor)

        world.tick()
            
        # Tick
        passed = False
        for step in tqdm.tqdm(range(max_timestep)):
            if COLLISION_STATE.all():
                passed = True
                break

            # apply controls
            [car.apply_control(carla.VehicleControl(throttle=1.0)) for car in car_list]

            # tick
            world.tick()

        if passed:
            print("PASS at {}".format(step))
        else:
            print("FAIL, exceeded {} steps".format(step))
            # print(car_name_list)
            # print(COLLISION_STATE)
            print([car_name_list[i] for i in range(num_cars) if COLLISION_STATE[i] == False])

if __name__ == "__main__":
    test_collision_sensor()
```

corkyw10, 2021-04-06:
@imoneoi Thank you for the input.

@DSantosO are you able to have a look at this please?

DSantosO, 2021-04-26:
Hello @imoneoi,
Thank you for reporting this. I have checked and there is an error in the collision settings of these vehicles. We will upload the fix soon.

DSantosO, 2021-08-04:
This error has been fixed in 0.9.12. 
Thanks again for reporting.

closed 2021-08-04 by DSantosO