carla-simulator/carla #4039: change duration of light state in traffic junction

logicalusmansherwani, 2021-03-30:
Hi

I simply load a map and run the ego vehicle using manual_control.py script with autopilot. The ego vehicle crosses the junction traffic signal. Currently the traffic light takes alot of time to set to green. I want to reduce the duration of traffic light state in the junction.

How can I change the duration of traffic light state. Is there any API which does this action.

I am using CARLA 0.9.11 on Fedora33

cyber-loop, 2021-03-31:
I think with set_green_time, set_yellow_time and set_red_time I can change the traffic light duration for all the traffic lights in the map. Can anyone confirm that?

https://carla.readthedocs.io/en/latest/python_api/#carlatrafficlight

closed 2021-03-31 by cyber-loop

PisanoCohen, 2022-05-31:
Hi, 

Did you manage to set the duration of each light states at a traffic light junction? I went through the docs and tried changing it directly through Python but is faced with some errors as shown below. Hope you can shed some light on this. 

[image]

I am using CARLA 0.9.12 on Windows 10. Python version 3.8.10

soumya997, 2022-11-28:
Check out this answer https://github.com/carla-simulator/carla/issues/876#issuecomment-1328671885