carla-simulator/carla #4043: Collision mode issue in co-simulation between CARLA & SUMO

EchoYJJ, 2021-03-31:
Hi! In the co-simulation, I use manual_control.py to interact with the vehicles generated in CARLA through spawn_npc_sumo.py.(named background vehicles **BV**)
However, the collision mode seems to be weird.
If the ego vehicle I control(named EV) slowly approaches the BV, it will detect me then yield. However, once EV collided with BV, the BV will become aggressive and bump EV like a bulldozer for the puropse to pass. Also, if EV rear-ends BV, only EV will slow down and no change seems to occur to BV.
I don't know why the collision mode seems to be unrealistic.
If you know the reason and the solution, please give me some advices.
Thanks in advance.

EchoYJJ, 2021-03-31:
Uploading 315485055e2c14ace73d5c83e5982d4e.mp4…

corkyw10, 2021-04-06:
@EchoYJJ thank you for the input.

@joel-mb Would you be able to have a look at this please?

EchoYJJ, 2021-04-07:
> @EchoYJJ thank you for the input.
> 
> @joel-mb Would you be able to have a look at this please?

Hi! I also notice that sometimes the BV will slow down when detecting EV, but sometimes it will just speed up regardless of collision.
I'm really curious about how the BV work in co-simulation between CARLA & SUMO(using run_synchronization.py or spawn_npc_sumo.py). Where can I find the source code of it, and whether it's easy to adjust.(python would be best)

Looking forward to your reply and many thanks.

joel-mb, 2021-04-07:
Hi @EchoYJJ, 
* *Realistic physics/collisions*: Right now, vehicles controlled by SUMO have physics disabled in CARLA, so collisions between CARLA and SUMO vehicles will never be realistic. Only vehicles controlled by CARLA have physics enabled.
*  *SUMO vehicles behaviors when colliding*: The behaviors you've seen are weird but I think this is more related to how SUMO handles this kind of situations. I would suggest taking a look to the different configurable parameters in SUMO (e.g., vtypes attributes https://sumo.dlr.de/docs/Definition_of_Vehicles%2C_Vehicle_Types%2C_and_Routes.html#available_vtype_attributes and SUMO simulation options https://sumo.dlr.de/docs/sumo.html#processing). Currently, the SUMO simulation is started with the following options: https://github.com/carla-simulator/carla/blob/master/Co-Simulation/Sumo/sumo_integration/sumo_simulation.py#L322

Regarding the source code for the co-simulation integration, you can find it here: https://github.com/carla-simulator/carla/tree/master/Co-Simulation/Sumo. Specifically, the core implementation is inside the `sumo_integration` folder.

Can you specify a little bit more about the `set_autopilot` issue? The `set_autopilot` is a method that you can use in CARLA to register/delete a vehicle from the CARLA traffic manager list (i.e., if autopilot is enabled, the vehicle is controlled by the CARLA traffic manager). Link to the PythonAPI documentation: https://carla.readthedocs.io/en/0.9.11/python_api/#setters_8

EchoYJJ, 2021-04-07:
Hi @joel-mb 
I wonder whether it's possible to enable the physics of vehicles controlled by SUMO, I think the interaction between CARLA vehicles and SUMO vehicles may be the biggest meaning of co-simulation.
If I get it right, when I run " run_synchronization.py or spawn_npc_sumo.py", the vehicles generated in CARLA are totally controlled by SUMO(name them background vehicles: BV). If I spawn a vehicle in CARLA by maunal_control.py(EV), it will spawn a corresponding vehicle in SUMO, and any influence on BV is generated in SUMO by SUMO logic. I don't know whether my understanding is correct.

The set_autopilot issue is just the way you understand, I wonder where I can find the full source core of it. In other word, I wonder how autopilot works in CARLA.

Many thanks!

joel-mb, 2021-04-12:
Hi @EchoYJJ !
* Exactly. You are right on how the co-simulation between CARLA and SUMO works. Basically, when you run the`run_synchronization` or `spawn_npc_sumo` scripts, all the vehicles spawned and controlled by SUMO are automatically generated in CARLA by the co-simulation package. It happens the same the other way around. I you spawn a vehicle in CARLA (e.g., using the `manual_control` script or the `spawn_npc` script), these vehicles are automatically generated in SUMO. Both simulations need to run in synchronous mode, and in every tick, the position and speed of each vehicle is updated in both simulations.
You can also take a look to https://drive.google.com/file/d/10DgMNUBqKqWBrdiwBiAIT4DdR9ObCquI/view if you  haven't done it yet.

* Regarding vehicle physics, this is a known limitation. You can try enabling physics in vehicles controlled by SUMO here: https://github.com/carla-simulator/carla/blob/master/Co-Simulation/Sumo/sumo_integration/carla_simulation.py#L113.

* Vehicles with `autopilot` enabled are controlled by the traffic manager. You can find all the related code here: https://github.com/carla-simulator/carla/tree/master/LibCarla/source/carla/trafficmanager. You can read about the internal design of the traffic manager here: https://carla.readthedocs.io/en/latest/adv_traffic_manager/

EchoYJJ, 2021-04-13:
Hey @joel-mb ! Many thanks for your detailed interpretation.
I change "carla.command.SetSimulatePhysics(carla.command.FutureActor, False))" from **False** to **True** as you suggest.
The situation seems to improve a little bit, but the vehicles controlled by SUMO will keep shaking and won't stop(or decelerate) when colliding with ego vehicle. The physics of them may not totally work.
https://user-images.githubusercontent.com/75726020/114488582-31137500-9c44-11eb-97d8-5fd9be48f255.mp4

As for the shaking issue, the step length firstly occurs to me. So, I set the --step--length from default 0.05 to 0.01. However, the simulation speed decline to a great extent. Do you know how to handle it?

Also, I notice that the simulation time in CARLA always runs faster than that of SUMO, but the co-simulation should be synchronized. Whether would this situation happens and which simulation time is correct?
https://user-images.githubusercontent.com/75726020/114488820-9f583780-9c44-11eb-896e-586d0d2bae8a.mp4

Many questions are asked, appreciate your help in advance!
Regards

joel-mb, 2021-04-14:
> As for the shaking issue, the step length firstly occurs to me. So, I set the --step--length from default 0.05 to 0.01. However, the  simulation speed decline to a great extent. Do you know how to handle it?

yeah, this is expected. When reducing the step length in synchronous mode the "speed" of the simulation will drop. This is because to simulate 1 second we need to tick more times to the server.

> Also, I notice that the simulation time in CARLA always runs faster than that of SUMO, but the co-simulation should be synchronized. Why would this situation happens and which simulation time is correct?

If the co-simulation is working in synchronous mode (this is mandatory) this should not happen. It may happen that the elapsed seconds since the simulation started is different between simulators, but once the co-simulation has started both simulations should run synchronously. Could you please specify with which parameter are you starting the co-simulation (e.g., step-length, ...). Also, did you do any other modification apart from enabling physics?

EchoYJJ, 2021-04-14:
Hi @joel-mb 
Sorry for my negligence, even if I didn't change anything in spawn_npc_sumo.py, I do shun the synchronous mode in run_synchronization.py before(for the test of speeding up simulation), and the latter is used by the former as a module.

However, even if SetSimulatePhysics set to be True, the vehicles generated in SUMO won't give a realistic physical feedback.(i.e. if the CARLA vehicle rear-ends a SUMO vehicle, CV will bounce while nothing happens to SV) , and I know that mainly because CARLA and SUMO will just judge the condition based on their own logic.
If the physical interaction can be that realistic, it will be really cool.
Anyway, thanks a lot for your patience!

closed 2023-06-02 by stale[bot]