carla-simulator/carla #4045: Height problem in co-simulation between SUMO and CARLA

EchoYJJ, 2021-03-31:
Hi! I'm trying to run a co-simulation between SUMO and CARLA in a overpass scene.
I use Roadrunner to raise my road into overpass.
However, if I use _python run_synchronization.py examples/osm.sumocfg  --sumo-gui_  to run the co-simulation, the vehicles generated by SUMO will only drive at horizontal plane(z=0) because the road in SUMO don't have height information.
So, in this situation, is that only option for me to generate vehicles in CARLA through spawn_npc_sumo.py?

corkyw10, 2021-04-06:
@EchoYJJ thank you for the question.

@joel-mb would you be able to follow up on this please?

joel-mb, 2021-04-07:
Hi @EchoYJJ, could you share the OpenDRIVE map you've generated with RoadRunner? Different road heights should be supported while running SUMO co-simulation.

EchoYJJ, 2021-04-08:
> Hi @EchoYJJ, could you share the OpenDRIVE map you've generated with RoadRunner? Different road heights should be supported while running SUMO co-simulation.

Hi! @joel-mb maybe I'm not making myself clear.
I generate vehicles in SUMO (defined in **rou** file), and the routes are based on the lanes (defined in **net** file). However the net file in SUMO doesn't contain any height information. As a result, even if I convert **net** to **xodr**, and adjust the height information in **xodr** by Roadrunner, the height changes won't reflect on **net**. So, when I run _run_synchronization.py_ , the vehicles generated in SUMO will just drive at horizontal plane(z=0) in CARLA like the video shows below.

https://user-images.githubusercontent.com/75726020/113958912-786cc080-9854-11eb-882e-a35315128af6.mp4

joel-mb, 2021-04-12:
@EchoYJJ So your initial SUMO net file doesn't contain information about road heights and you use RoadRunner to include this information in the OpenDrive? If this is your pipeline, you need to update also the SUMO net to include the road height information from your modified OpenDrive file. To do this you can use the `netconvert_carla` script (https://github.com/carla-simulator/carla/blob/master/Co-Simulation/Sumo/util/netconvert_carla.py). Internally, this script uses `netconvert` to generate
the net and inserts, manually, the traffic light landmarks retrieved from the opendrive.

On the other hand, you can also use the `spawn_npc_sumo` script. This script automatically generates a SUMO net based on the OpenDrive map loaded in CARLA.

EchoYJJ, 2021-04-14:
Hi @joel-mb couldn't thank you any more! 
netconvert_carla is such a strong tool that I wish I had known about it earlier.
I have one last confusion. By doing so, the shape information in net file seems to add another dimension(**-0.00&0.04** in bold).
_shape="442.29,752.15,**-0.00** 442.28,751.15,**0.04**"_
Is that dimension gradient or height?

joel-mb, 2021-04-14:
Hi @EchoYJJ. I would expect this to be the height.

closed 2021-04-14 by EchoYJJ