carla-simulator/carla #4049: how can I add a driver or passenger into the car

Guocode, 2021-04-01:
I would like to simulate a incarbin view with driver and passengers, is there any easy method to realize it or I must customize my own car model with passengers and driver and add it as a vehicle?

corkyw10, 2021-04-06:
@Guocode thank you for the question.

@gsmario would you be able to follow up on this please?

gsmario, 2021-04-06:
Hello, first thank you for your question! 
You'll need 3 important things in order to do this,  the car, the driver/etc and the animation/pose for the human.
If you just want to add a driver/passenger in to the vehicle, you need to pose/animate the human in order to fit inside your vehicle like it should. You can do this using any 3d modeling software.

Once you've posed your driver, you can import it to unreal engine, and add the skeletal mesh inside the vehicle blueprint, (exactly as the other car parts, doors, glasses. 
Or you can even export the human model together with the car, if you like.

Guocode, 2021-04-07:
@gsmario Thanks for reply, I will try as you say. I would be pleasure if this can be a feature of carla, because it will be more real that there is a driver in a heading on car.

closed 2021-04-07 by Guocode

seuwcs, 2021-04-23:
@gsmario  Sorry for disturbing. I'd like to know if the first view of drivers can be seen from carla?