carla-simulator/carla #4051: Docker multistage build.

cinjon, 2021-04-01:
I am using Carla in Docker. The following works well:

```
FROM carlasim/carla:0.9.11 as carlasim

USER root
RUN apt-get update
RUN apt-get install -y software-properties-common &&\
        apt-add-repository universe &&\
            apt-get update &&\
            apt-get install -y python3-pip
RUN apt-get install -y libpng16-16 libtiff5 libjpeg-turbo8 wget && rm -rf /var/lib/apt/lists/*
RUN apt-get update

RUN chmod -R 777 .

ENV CARLAPATH /home/carla/PythonAPI/carla/
USER carla
```

However, I need to use a multistage build (with pytorch) and when I include that after this one, then I run into the issues around xdg-user-dir referenced multiple times (e.g. #1892 and #225). Is there some one liner that I'm missing here? The following is the extended multi-stage Dockerfile:

```
FROM carlasim/carla:0.9.11 as carlasim

USER root
RUN apt-get update
RUN apt-get install -y software-properties-common &&\
        apt-add-repository universe &&\
            apt-get update &&\
            apt-get install -y python3-pip
RUN apt-get install -y libpng16-16 libtiff5 libjpeg-turbo8 wget && rm -rf /var/lib/apt/lists/*
RUN apt-get update

RUN chmod -R 777 .

ENV CARLAPATH /home/carla/PythonAPI/carla/
USER carla

FROM nvcr.io/nvidia/pytorch:20.12-py3

COPY --from=carlasim /home/carla /home/carla

USER root
RUN useradd -ms /bin/bash me

RUN apt-get update
ARG DEBIAN_FRONTEND=noninteractive
ENV TZ=America/New_York
RUN apt-get install libgl1-mesa-glx -y
RUN apt-get install ffmpeg libsm6 libxext6  -y
RUN apt-get install -y software-properties-common &&\
        apt-add-repository universe &&\
            apt-get update &&\
            apt-get install -y python3-pip
RUN apt-get install -y libpng16-16 libtiff5 libjpeg-turbo8 wget && rm -rf /var/lib/apt/lists/*
RUN apt-get update

WORKDIR /home/me

RUN wget \
    https://repo.anaconda.com/miniconda/Miniconda3-latest-Linux-x86_64.sh \
    # && mkdir ./.conda \
    && bash Miniconda3-latest-Linux-x86_64.sh -b \
    && rm -f Miniconda3-latest-Linux-x86_64.sh

RUN chmod -R 777 ./

USER me

ENV PATH="./miniconda3/bin:${PATH}"
ARG PATH="./miniconda3/bin:${PATH}"

RUN conda create -n myenv python=3.7.9

SHELL ["conda", "run", "-n", "myenv", "/bin/bash", "-c"]

RUN conda install pytorch torchvision torchaudio cudatoolkit=11.1 -c pytorch -c conda-forge

RUN pip install --upgrade pip
RUN pip install nuscenes-devkit
RUN pip install pygame networkx

ENV CARLAPATH=/home/carla/PythonAPI/carla/
RUN echo ${CARLAPATH}

COPY . /home/me
WORKDIR /home/me

RUN pip install --no-cache-dir -r requirements.txt
```

qhaas, 2021-04-02:
As mentioned in Issue #3514 , add the `xdg-user-dirs xdg-utils` packages to the `apt-get install` line if your final image.

cinjon, 2021-04-02:
I apologize but I wasn't clear enough. This is actually causing my build to not work at all. It's not that there's an errant warning - it's actually just exiting from Carla immediately after running the command without much insight into why.

I did just do what you suggested and the warning was gone, but Carla still doesn't work.

```
(myenv) me@336dcd2d6b5c:~$ SDL_VIDEODRIVER=offscreen SDL_HINT_CUDA_DEVICE=2 /home/carla/CarlaUE4.sh -quality-level=Epic -opengl
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
````

Below is my new Dockerfile with those changes. I am running this on the exact same machine where, outside of Docker, I can run the command np.

```
FROM carlasim/carla:0.9.11 as carlasim

USER root
RUN apt-get update
RUN apt-get install -y software-properties-common &&\
        apt-add-repository universe &&\
            apt-get update &&\
            apt-get install -y python3-pip
RUN apt-get install -y libpng16-16 libtiff5 libjpeg-turbo8 wget && rm -rf /var/lib/apt/lists/*
RUN apt-get update

RUN chmod -R 777 .

ENV CARLAPATH /home/carla/PythonAPI/carla/
USER carla

FROM nvcr.io/nvidia/pytorch:20.12-py3

COPY --from=carlasim /home/carla /home/carla

USER root
RUN useradd -ms /bin/bash me

RUN apt-get update
ARG DEBIAN_FRONTEND=noninteractive
ENV TZ=America/New_York
RUN apt-get install libgl1-mesa-glx -y
RUN apt-get install ffmpeg libsm6 libxext6  -y
RUN apt-get install -y software-properties-common &&\
        apt-add-repository universe &&\
            apt-get update &&\
            apt-get install -y python3-pip
RUN apt-get install -y libpng16-16 libtiff5 libjpeg-turbo8 wget && rm -rf /var/lib/apt/lists/*
RUN apt-get update

RUN apt-get install -y xdg-user-dirs xdg-utils
RUN apt-get update

WORKDIR /home/me

RUN wget \
    https://repo.anaconda.com/miniconda/Miniconda3-latest-Linux-x86_64.sh \
    # && mkdir ./.conda \
    && bash Miniconda3-latest-Linux-x86_64.sh -b \
    && rm -f Miniconda3-latest-Linux-x86_64.sh

RUN chmod -R 777 ./

USER me

ENV PATH="./miniconda3/bin:${PATH}"
ARG PATH="./miniconda3/bin:${PATH}"

RUN conda create -n myenv python=3.7.9

SHELL ["conda", "run", "-n", "myenv", "/bin/bash", "-c"]

RUN conda install pytorch torchvision torchaudio cudatoolkit=11.1 -c pytorch -c conda-forge

RUN pip install --upgrade pip
RUN pip install nuscenes-devkit
RUN pip install pygame networkx

ENV CARLAPATH=/home/carla/PythonAPI/carla/
RUN echo ${CARLAPATH}
ENV SDL_VIDEODRIVER=offscreen

COPY . /home/me
WORKDIR /home/me

RUN pip install --no-cache-dir -r requirements.txt
```

qhaas, 2021-04-02:
> Carla still doesn't work

Upon closer inspection of your dockerfile, it likely lacks everything you need to run OpenGL/Vulkan:  `FROM nvcr.io/nvidia/pytorch:20.12-py3`

I doubt a pytorch container will have all the opengl/vulkan dependencies.  Recommend deploying the dependencies noted in the [nvidia opengl container](https://gitlab.com/nvidia/container-images/opengl/-/blob/ubuntu18.04/glvnd/runtime/Dockerfile) and/or the [nvidia vulkan container](https://gitlab.com/nvidia/container-images/vulkan/-/blob/master/docker/Dockerfile.ubuntu18.04)

cinjon, 2021-04-03:
I appreciate that pointer. I modified the Dockerfile to include the opengl additions, with amd64 instead of i386 (see below). I can now launch it with `SDL_HINT_CUDA_DEVICE=2 /home/carla/CarlaUE4.sh -opengl 2>&1 > /dev/null &` but when I try to have my client call the server, which is in the same docker, I get this:

```
$ docker run --runtime=nvidia --net=host -e NVIDIA_VISIBLE_DEVICES=2,3 -it mydocker:mytag bash
$ (/home/carla/CarlaUE4.sh -opengl 2>&1 > /dev/null &) && (sleep 15)
$ ps aux | grep carla
foo     222  0.0  0.0   2616   552 pts/0    S    20:53   0:00 /bin/dash /home/carla/CarlaUE4.sh -opengl
foo     230  227  0.8 8466904 2323656 pts/0 Sl   20:53   1:01 /home/carla/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping CarlaUE4 -opengl
$ python main.py --gpu 1
pygame 2.0.1 (SDL 2.0.14, Python 3.7.9)
Hello from the pygame community. https://www.pygame.org/contribute.html
LowLevelFatalError [File:Unknown] [Line: 1155]
GameThread timed out waiting for RenderThread after 60.00 secs
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554
Malloc Size=65535 LargeMemoryPoolOffset=131119
Malloc Size=137456 LargeMemoryPoolOffset=268592
Segmentation fault (core dumped)
Traceback (most recent call last):
  File "main.py", line 436, in <module>
    run(opts)
  File "main.py", line 63, in run
    client.load_world(opts.map_name)
RuntimeError: time-out of 30000ms while waiting for the simulator, make sure the simulator is ready and connected to 127.0.0.1:2000
$ ps aux | grep carla
foo     452  0.0  0.0   5188   736 pts/0    S+   20:56   0:00 grep --color=auto carla
```

So somewhere in there the simulator just died, seemingly from a memory dump. Both of the visible gpus have plenty of memory available and neither loaded anything. 

```
FROM carlasim/carla:0.9.11 as carlasim

USER root
RUN apt-get update
RUN apt-get install -y software-properties-common &&\
        apt-add-repository universe &&\
            apt-get update &&\
            apt-get install -y python3-pip
RUN apt-get install -y libpng16-16 libtiff5 libjpeg-turbo8 wget && rm -rf /var/lib/apt/lists/*
RUN apt-get update

RUN chmod -R 777 .

ENV CARLAPATH /home/carla/PythonAPI/carla/
USER carla

FROM nvcr.io/nvidia/pytorch:20.12-py3

COPY --from=carlasim /home/carla /home/carla

USER root
RUN useradd -ms /bin/bash me

RUN apt-get update
ARG DEBIAN_FRONTEND=noninteractive
ENV TZ=America/New_York
RUN apt-get install libgl1-mesa-glx -y
RUN apt-get install ffmpeg libsm6 libxext6  -y
RUN apt-get install -y software-properties-common &&\
        apt-add-repository universe &&\
            apt-get update &&\
            apt-get install -y python3-pip
RUN apt-get install -y libpng16-16 libtiff5 libjpeg-turbo8 wget && rm -rf /var/lib/apt/lists/*

RUN dpkg --add-architecture amd64
RUN apt-get update
RUN apt-get update && apt-get install -y --no-install-recommends \
        libglvnd0 libglvnd0:amd64 \
        libgl1 libgl1:amd64 \
        libglx0 libglx0:amd64 \
        libegl1 libegl1:amd64 \
        libgles2 libgles2:amd64 && \
    rm -rf /var/lib/apt/lists/*
COPY dockerfiles/10_nvidia.json /usr/share/glvnd/egl_vendor.d/10_nvidia.json

RUN apt-get update
RUN apt-get install -y xdg-user-dirs xdg-utils
RUN apt-get update

WORKDIR /home/me

RUN wget \
    https://repo.anaconda.com/miniconda/Miniconda3-latest-Linux-x86_64.sh \
    # && mkdir ./.conda \
    && bash Miniconda3-latest-Linux-x86_64.sh -b \
    && rm -f Miniconda3-latest-Linux-x86_64.sh 

RUN chmod -R 777 ./

USER me

ENV PATH="./miniconda3/bin:${PATH}"
ARG PATH="./miniconda3/bin:${PATH}"

RUN conda create -n myenv python=3.7.9

SHELL ["conda", "run", "-n", "myenv", "/bin/bash", "-c"]

RUN python -V

RUN conda install pytorch torchvision torchaudio cudatoolkit=11.1 -c pytorch -c conda-forge

RUN pip install --upgrade pip
RUN pip install nuscenes-devkit
RUN pip install pygame networkx

ENV CARLAPATH=/home/carla/PythonAPI/carla/
ENV SDL_VIDEODRIVER=offscreen
RUN echo ${CARLAPATH}
RUN echo ${SDL_VIDEODRIVER}

COPY . /home/me
WORKDIR /home/me

RUN pip install torch-scatter -f https://pytorch-geometric.com/whl/torch-1.8.0+cu111.html
RUN pip install torch-sparse -f https://pytorch-geometric.com/whl/torch-1.8.0+cu111.html
RUN pip install torch-cluster -f https://pytorch-geometric.com/whl/torch-1.8.0+cu111.html
RUN pip install torch-spline-conv -f https://pytorch-geometric.com/whl/torch-1.8.0+cu111.html
RUN pip install torch-geometric

RUN pip install --no-cache-dir -r requirements.txt

COPY dockerfiles/asound.conf /etc/asound.conf
```

Further help is super appreciated.

cinjon, 2021-04-03:
Does this possibly have to do with me using Ubuntu 20.04 from the pytorch image? Asking because I came across this article from Nov '20 that suggests that Carla doesn't work with Ubuntu 20.04. Can't tell if that's correct still today though.

cinjon, 2021-04-03:
That doesn't seem to have been it. I just tried to do everything the same but use `nvcr.io/nvidia/pytorch:20.11-py3`, which is 18.04 (verified) and I got the following error both with and without the `DISPLAY=`

```
$ (DISPLAY= /home/carla/CarlaUE4.sh -opengl 2>&1 > /dev/null &) && (sleep 15)
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554
Malloc Size=65535 LargeMemoryPoolOffset=131119
Malloc Size=106784 LargeMemoryPoolOffset=237920
Segmentation fault (core dumped)
```

qhaas, 2021-04-03:
The pytorch container likely doesn't have the same [driver capabilities](https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/user-guide.html#driver-capabilities) set as the carla container since it is designed for compute, not rendering (I did not verify), you will need to make sure `graphics` is in your customized container:
```
$ podman run --rm -it docker.io/carlasim/carla:0.9.11 printenv | grep NVIDIA_DRIVER_CAPABILITIES
NVIDIA_DRIVER_CAPABILITIES=compute,graphics,utility
```

I'd also start with a simpler, non-background run command inside that container (albeit the one you posted 'should' work), try starting it with just `DISPLAY= CarlaUE4.sh -opengl` and see if it stays up.

While it likely isn't the problem, you are using the 'classic' approach to starting the container with GPU support, not sure if newer versions of nvidia-docker2 work properly with that syntax to select GPUs, see [example using specific GPUs](https://docs.nvidia.com/deeplearning/frameworks/user-guide/index.html#enable-gpu-support):  `docker run --gpus "device=2,3"`

Alternative 1:
You don't need multi-stage builds here (albeit, I don't think multi-stage is causing the problem), just install the Ubuntu CARLA deb package into your pytorch container using [the CARLA apt repo](https://carla.readthedocs.io/en/latest/start_quickstart/#a-debian-carla-installation).

Alternative 2:
The server/client need not run in the same container...  You can install a CARLA client into a python environment inside the NGC PyTorch container and leave CARLA server running in an unmodified (i.e. direct from dockerhub) CARLA container, which appears to run fine on your system.  You do not need GPU support for the CARLA client.  This is an advantage of CARLA's [server-client architecture](https://carla.readthedocs.io/en/0.9.8/start_introduction/#the-simulator).  You'd just need to use some of docker's [networking capabilities](https://docs.docker.com/network/) to setup an isolated, local network for the two containers and have the hostnames of the two containers resolve.  One 'service' per container is [encouraged by docker](https://docs.docker.com/config/containers/multi-service_container/).

cinjon, 2021-04-05:
Thanks for continuing to help. I really do appreciate it.

Re alternative 1, which was just doing the pytorch package with a Carla apt repo installation, I've tried a bunch of stuff and I can't get past just a segfault immediately upon loading. However, this time I got an actual error response, which I see was brought up recently in #2826. It can't be a GPU memory issue though (as posited in the aforementioned issue) because I am using a Titan V and set it so that was the only visible one.

```
me@e7e57d8d71da:~$ DISPLAY= /opt/carla-simulator/CarlaUE4.sh -quality-level=Epic -opengl
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=106784 LargeMemoryPoolOffset=237920 
Failed to find symbol file, expected location:
"/opt/carla-simulator/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping.sym"
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
```

Below I put the example Dockerfile I used. Following the above, I then tried to substitute the carla version, i.e. `RUN apt-get install carla-simulator -y` instead of `RUN apt-get install carla-simulator=0.9.11 -y`, but got the exact same error:

```
me@0e1acc5361c1:~$ DISPLAY= /opt/carla-simulator/CarlaUE4.sh -quality-level=Epic -opengl
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=106784 LargeMemoryPoolOffset=237920 
Failed to find symbol file, expected location:
"/opt/carla-simulator/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping.sym"
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
```

Dockerfile:

```
FROM nvcr.io/nvidia/pytorch:20.11-py3

USER root
RUN useradd -ms /bin/bash me

RUN apt-get update
ARG DEBIAN_FRONTEND=noninteractive
ENV TZ=America/New_York
RUN apt-get install libgl1-mesa-glx -y
RUN apt-get install ffmpeg libsm6 libxext6  -y
RUN apt-get install -y software-properties-common &&\
        apt-add-repository universe &&\
            apt-get update &&\
            apt-get install -y python3-pip
RUN apt-get install -y libpng16-16 libtiff5 libjpeg-turbo8 wget && rm -rf /var/lib/apt/lists/*

RUN dpkg --add-architecture amd64
RUN apt-get update
RUN apt-get update && apt-get install -y --no-install-recommends \
        libglvnd0 libglvnd0:amd64 \
        libgl1 libgl1:amd64 \
        libglx0 libglx0:amd64 \
        libegl1 libegl1:amd64 \
        libgles2 libgles2:amd64 && \
    rm -rf /var/lib/apt/lists/*
COPY dockerfiles/10_nvidia.json /usr/share/glvnd/egl_vendor.d/10_nvidia.json

RUN apt-get update
RUN apt-get install -y xdg-user-dirs xdg-utils

RUN apt-get update
RUN apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 1AF1527DE64CB8D9
RUN add-apt-repository "deb [arch=amd64] http://dist.carla.org/carla $(lsb_release -sc) main"
RUN apt-get update # Update the Debian package index
RUN apt-get install carla-simulator=0.9.11 -y

WORKDIR /home/me

RUN wget \
    https://repo.anaconda.com/miniconda/Miniconda3-latest-Linux-x86_64.sh \
    # && mkdir ./.conda \
    && bash Miniconda3-latest-Linux-x86_64.sh -b \
    && rm -f Miniconda3-latest-Linux-x86_64.sh 

RUN chmod -R 777 ./

USER me

ENV PATH="./miniconda3/bin:${PATH}"
ARG PATH="./miniconda3/bin:${PATH}"

RUN conda create -n myenv python=3.7.9

SHELL ["conda", "run", "-n", "myenv", "/bin/bash", "-c"]

RUN python -V

RUN conda install pytorch torchvision torchaudio cudatoolkit=11.1 -c pytorch -c conda-forge

RUN pip install --upgrade pip
RUN pip install nuscenes-devkit
RUN pip install pygame networkx

ENV CARLAPATH=/opt/carla-simulator/PythonAPI/carla/
ENV SDL_VIDEODRIVER=offscreen
RUN echo ${CARLAPATH}
RUN echo ${SDL_VIDEODRIVER}

COPY . /home/me
WORKDIR /home/me

RUN pip install torch-scatter -f https://pytorch-geometric.com/whl/torch-1.8.0+cu111.html
RUN pip install torch-sparse -f https://pytorch-geometric.com/whl/torch-1.8.0+cu111.html
RUN pip install torch-cluster -f https://pytorch-geometric.com/whl/torch-1.8.0+cu111.html
RUN pip install torch-spline-conv -f https://pytorch-geometric.com/whl/torch-1.8.0+cu111.html
RUN pip install torch-geometric

RUN pip install --no-cache-dir -r requirements.txt

COPY dockerfiles/asound.conf /etc/asound.conf
```

cinjon, 2021-04-08:
I think this is solved now.
The approach that worked was doing the pytorch image, apt-get the carla install, and then ENSURING that the NVIDIA_DRIVER_CAPABILITIES env var was set to include graphics. Thanks much for your help @qhaas .

closed 2021-04-08 by cinjon

DeathlyMade, 2024-10-14:
Can you send the final dockerfile that worked?