carla-simulator/carla #4052: Issue on CARLA 0.9.11 bulid on Windows.(fatal: repository 'https://github.com/EpicGames/UnrealEngine.git/' not found)

EchoYJJ, 2021-04-02:
Hi! I'm trying to build CARLA 0.9.11 on Windows.
I already link my GitHub account  to Unreal Engine's account.
[image]

However, when I run "git clone --depth=1 -b 4.24 https://github.com/EpicGames/UnrealEngine.git" and sign in my account like below.
[image]

I still receive the warning:
_PS F:\Unreal Engine> git clone --depth=1 -b 4.24 https://github.com/EpicGames/UnrealEngine.git
Cloning into 'UnrealEngine'...
info: please complete authentication in your browser...
remote: Repository not found.
fatal: repository 'https://github.com/EpicGames/UnrealEngine.git/' not found_

If you know how to solve this issue, please give me some advices, thanks in advance.

EchoYJJ, 2021-04-02:
[image]
I've found out the solution, must become a member of it.

closed 2021-04-02 by EchoYJJ

vibhoraggarwal, 2021-05-05:
For someone following this issue: There would be a link on the email id attached with the github profile with the link to become a member of the Epic Games community