carla-simulator/carla #4053: Update CARLA docker runtime documentation

qhaas, 2021-04-02:
The existing [CARLA runtime documentation](https://carla.readthedocs.io/en/latest/build_docker/#running-carla-container) is quite dated and conflicts with the guidance provided in the latest (0.9.11) [runtime Dockerfile](https://github.com/carla-simulator/carla/blob/0.9.11/Util/Docker/Release.Dockerfile).  It would likely reduce confusion to have the former reflect the latter.

corkyw10, 2021-04-06:
Hi @qhaas thank you for the input. I will review the documentation and update it.

kino3, 2021-07-04:
@qhaas @corkyw10 Hi, I'm trying to use Carla runtime by the docker image but not work well. As you mentioned above, [the documentation ](https://carla.readthedocs.io/en/latest/build_docker/#running-carla-container) is outdated and [the comments in Dockerfile](https://github.com/carla-simulator/carla/blob/0.9.11/Util/Docker/Release.Dockerfile) is true?

corkyw10, 2021-07-07:
Hi @kino3 the documentation is being updated but is not yet ready. The dockerfile you referenced is used to build the current carla docker image. What issue specifically are you facing?

kino3, 2021-07-11:
@corkyw10 Thank you for your reply. Now my environment works well by using the command `docker run   -e SDL_VIDEODRIVER=x11   -e DISPLAY=$DISPLAY  -v /tmp/.X11-unix:/tmp/.X11-unix   -p 2000-2002:2000-2002   -it   --gpus all carlasim/carla:0.9.11 ./CarlaUE4.sh -opengl` 
ISSUE #3444 #3431 helped me well.

PDillis, 2021-07-11:
@kino3 I also managed to run it with the command:

```bash
docker run --rm -p 2000-2002:2000-2002 --gpus 'device=0' carlasim/carla:0.9.11 /bin/bash -c 'SDL_VIDEODRIVER=offscreen ./CarlaUE4.sh -nosound'
```
Perhaps this is a simpler version? Let me know if it also works for you.

corkyw10, 2021-08-05:
Docker documentation has been updated in the latest release [here](https://carla.readthedocs.io/en/0.9.12/build_docker/) and [here](https://carla.readthedocs.io/en/0.9.12/build_docker_unreal/). If you see any problems please open a new github issue.

closed 2021-08-05 by corkyw10