carla-simulator/carla #4054: How to get imu and gnss datas as csv file

AydinAltun, 2021-04-02:


AydinAltun, 2021-04-02:
Hi , i wanna get the data from imu and gnss sensors for sensor fusion  and i found a script from the internet  but it did not work, is there a way to save these datas to a csv file ?

corkyw10, 2021-04-06:
Hi @AydinAltun previous users have used the [csv python module](https://docs.python.org/3/library/csv.html), such as in this [issue](https://github.com/carla-simulator/carla/issues/2218). Does this help you?

AydinAltun, 2021-04-06:
Actually it partly worked because i could save the imu datas to csv file but when i want to save gnss values it gives error which is
**  File "save.py", line 1027, in game_loop
    "latitude":world.gnss_sensor.latitude[0],
AttributeError: 'GnssSensor' object has no attribute 'latitude'
Fatal Python error: PyEval_SaveThread: NULL tstate
**

Here is the part of recording codes:
low_data = {"accelX":world.imu_sensor.accelerometer[0],
                        "accelY":world.imu_sensor.accelerometer[1],
                        "accelZ":world.imu_sensor.accelerometer[2],
                        "gyroX":world.imu_sensor.gyroscope[0], 
                        "gyroY":world.imu_sensor.gyroscope[1], 
                        "gyroZ":world.imu_sensor.gyroscope[2],
                        
                        "latitude":world.gnss_sensor.latitude[0],
                        "longitude":world.gnss_sensor.longitude[0],
                        "altitude":world.gnss_sensor.altitude[0],
                        "class":str(args.name)}
            data = data.append(low_data, ignore_index=True)

corkyw10, 2021-04-08:
Can you share how you are spawning the sensors?

AydinAltun, 2021-04-08:
"""
    W            : throttle
    S            : brake
    A/D          : steer left/right
    Q            : toggle reverse
    Space        : hand-brake
    P            : toggle autopilot
    M            : toggle manual transmission
    ,/.          : gear up/down

    L            : toggle next light type
    SHIFT + L    : toggle high beam
    Z/X          : toggle right/left blinker
    I            : toggle interior light

    TAB          : change sensor position
    ` or N       : next sensor
    [1-9]        : change to sensor [1-9]
    G            : toggle radar visualization
    C            : change weather (Shift+C reverse)
    Backspace    : change vehicle

    R            : toggle recording images to disk

    CTRL + R     : toggle recording of simulation (replacing any previous)
    CTRL + P     : start replaying last recorded simulation
    CTRL + +     : increments the start time of the replay by 1 second (+SHIFT = 10 seconds)
    CTRL + -     : decrements the start time of the replay by 1 second (+SHIFT = 10 seconds)

    F1           : toggle HUD
    H/?          : toggle help
    ESC          : quit
"""

from __future__ import print_function

# ==============================================================================
# -- find carla module ---------------------------------------------------------
# ==============================================================================

import pandas as pd
import glob
import os
import sys

try:
    sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    pass

# ==============================================================================
# -- imports -------------------------------------------------------------------
# ==============================================================================

import carla

from carla import ColorConverter as cc

import argparse
import collections
import datetime
import logging
import math
import random
import re
import weakref

[... 978 lines cut ...]
        '--name',
        default="Aydin",
        type=str,
        help='Name Of Player')
    args = argparser.parse_args()

    args.width, args.height = [int(x) for x in args.res.split('x')]

    log_level = logging.DEBUG if args.debug else logging.INFO
    logging.basicConfig(format='%(levelname)s: %(message)s', level=log_level)

    logging.info('listening to server %s:%s', args.host, args.port)

    print(__doc__)

    try:

        game_loop(args)

    except KeyboardInterrupt:
        print('\nCancelled by user. Bye!')

if __name__ == '__main__':

    main()

AydinAltun, 2021-04-08:
https://github.com/AydinAltun/Carla-get-datas/blob/main/record.py
it is how i tried to get the sensors datas

corkyw10, 2021-04-09:
Ok, it's because of how the Sensor is being called in this script. If you look at line [750](https://github.com/AydinAltun/Carla-get-datas/blob/main/record.py#L750) you will see that the latitude and longitude are being asigned to values `lat` and `long`. So I believe you will need to change your code after line 750 as follows to get the altitude as well:

```
    @staticmethod
    def _on_gnss_event(weak_self, event):
        self = weak_self()
        if not self:
            return
        self.lat = event.latitude
        self.lon = event.longitude
        self.alt = event.altitude
```

And to get the values change after line 1003:

```
            low_data = {"accelX":world.imu_sensor.accelerometer[0],
                        "accelY":world.imu_sensor.accelerometer[1],
                        "accelZ":world.imu_sensor.accelerometer[2],
                        "gyroX":world.imu_sensor.gyroscope[0], 
                        "gyroY":world.imu_sensor.gyroscope[1], 
                        "gyroZ":world.imu_sensor.gyroscope[2],
                        
                        "latitude":world.gnss_sensor.lat,
                        "longitude":world.gnss_sensor.lon,
                        "altitude":world.gnss_sensor.alt,
                        "class":str(args.name)}
```

AydinAltun, 2021-04-09:
Thanks so much , i realized that

closed 2021-04-09 by corkyw10