carla-simulator/carla #4056: New logitech g29 steering issue that vehicle stay still not matter which pedal I pressed

seuwcs, 2021-04-03:
Dear all friends,
 I am connecting the logitech g29 to carla (version 0.9.10) on Linux Ubuntu 18.04, but something went wrong after the connection was made. My confusion is that: the car generated in the scene by the script "manual_control_steeringwheel.py " could not be controlled by the steering wheel as  well as the pedal.

 The vehicle remains still not matter which pedal I stepped, the instruction said that I should firstly press the brake pedal, but it doesn't work, the value of the brake pedal remain unchanged . I also carefully read the issue (https://github.com/carla-simulator/carla/issues/1521), but my question could not be solved at all, which confused me a lot, who can give me some suggestions? any advice will be appreciated greatly.

The picture of the scene is as follows (in the redrectangle ):
[image]

thank brothers !
@germanros1987

corkyw10, 2021-04-06:
@seuwcs Thank you for the question.

@germanros1987 are you able to follow up on this?

GeraldGuo7, 2021-05-10:
"manual_control_steeringwheel.py " this file need to be modified for the logitechg29 and you need to add the config.ini.You can find some tutorials in the internet.
I just got into the shallow part about the Carla simulator and luckily I solved this issue with Carla version 0.9.11.

kaka-lin, 2022-05-27:
Hi, I also got the same issue

debipe20, 2023-06-22:
Could you provide any instruction on how to setup Logitech Steering wheel and pedal with CARLA?

closed 2026-11-13 by carla-triage[bot]