carla-simulator/carla #4057: Pedestrian navigation binary very small & failing

JulianPetzold, 2021-04-03:
Hi!
For use in CARLA 9.11 built from source on Ubuntu 18.04, I built a custom map with RoadRunner consisting of a few buildings and an 8-shaped road with a crosswalk at one point. 

After loading the map for the first time with "make import" I could make adjustments in the UE4 editor and then exported the map according to the tutorial (Tagging the sky box, renaming the roads, adding a crosswalk mesh, ...).
I generated the binary file for the pedestrian navigation using the build.sh script and put it into the "Nav" subfolder of the map. 
Curiously, this folder was not already present like in the tutorial I followed and I had to create it. Furthermore, the binary only has a size of 7.7KB, which seems to be smaller than the tutorial as well.

Unfortunately, this pedestrian navigation file does not seem to work and I cannot determine why. When I try to use the example script "spawn_npc.py", only vehicles are spawned but no pedestrians. When I forcefully spawn a pedestrian with my own script and try to activate the automatic WalkerAIController with the `start()` method, the script crashes with a segmentation fault. 

Does anybody know what the root cause of this could be?

corkyw10, 2021-04-06:
Hi @JulianPetzold, there is an updated version of that tutorial in the development/latest version of the [docs](https://carla.readthedocs.io/en/latest/tuto_A_add_map/add_map_source/#generate-pedestrian-navigation). Often the problem is that not all the correct meshes were selected before exporting to create the `.obj` file. After tagging large meshes not needed for pedestrian navigation with `noExport`, select all the meshes with `ctrl + A` (if you are following the old tutorial it says to press `G` and this seems to create problems) and then select `File -> Carla Exporter` to export the `.obj`.  Then go to the `.obj` file and open it to see if the right meshes have been exported. Search for the mesh names, e.g., `Road_Sidewalk` to make sure they were exported. Then move the `<mapName>.obj` and the `<mapName>.xodr` to `Util/DockerUtils/dist` and then run `./build.sh <mapName>`.

Let me know if that helps

JulianPetzold, 2021-04-19:
Hi @corkyw10, thank you for your reply, the new method works for us.

closed 2021-04-19 by JulianPetzold