carla-simulator/carla #4059: Visual Studio Community 2017 installation problems

huobinzy, 2021-04-04:
Hello!
When I install the VS according to the following instructions, VS installer has checked some components by default. Do I need to cancel the default ones?
Especially Desktop development with C++, which checked the Windows 10 SDK (10.0.17763.0). when I made PythonAPI, a part of the reminder was not using the Windows 8.1 SDK, but the Windows 10 SDK (10.0.17763.0). So I guess the fatal failure may be caused by this SDK misuse.
Can you give me clearer and more detailed instructions based on following instructions.
Thank you very much！

1. Windows 8.1 SDK. Select it in the Installation details section on the right or go to the Indivdual Components tab and look under the SDKs, libraries, and frameworks heading.
2. x64 Visual C++ Toolset. In the Workloads section, choose Desktop development with C++. This will enable a x64 command prompt that will be used for the build. Check that it has been installed correctly by pressing the Windows button and searching for x64. Be careful not to open a x86_x64 prompt.
3. .NET framework 4.6.2. In the Workloads section, choose .NET desktop development and then in the Installation details panel on the right, select .NET Framework 4.6.2 development tools. This is required to build Unreal Engine.

corkyw10, 2021-04-06:
@bernatx could you follow up on this please?

huobinzy, 2021-04-07:
> @bernatx could you follow up on this please?

Aha, thank you for conernatx could you follow up on this please?

Aha, thank you for con

closed 2021-04-07 by huobinzy

BuBleC, 2022-03-09:
@huobinzy Hello I currently have the same problem could you help me? I don't have a good understanding of GitHub so what does add fodler to repository and new commit mean?