carla-simulator/carla #4060: Make PythonAPI failing on Windows(CARLA 0.9.11): failed with exit status 1120

EchoYJJ, 2021-04-04:
Hi! I'm trying to build CARLA 0.9.11 on windows.
After running "make PythonAPI", my [Setup] receive success. However, my [BuildLibCarla] ends up with warning below:
_build\lib.win-amd64-3.8\carla\libcarla.cp38-win_amd64.pyd : fatal error LNK1120: 77 个无法解析的外部命令(unresolvable external commands)
error: command 'D:\\VS2017\\VC\\Tools\\MSVC\\14.16.27023\\bin\\HostX86\\x64\\link.exe' failed with exit status 1120_

Although it says _"-[BuildPythonAPI]: Carla lib for python has been successfully installed in "D:\carla\PythonAPI\carla\dist"!"_,
dist doesn't show up in carla.

I also notice many warnings shows up like below:
libcarla.obj : error LNK2001: 无法解析的外部符号(unresolvable external commands) "class boost::python::handle<struct _typeobject> __cdecl boost::python::objects::registered_class_object(struct boost::python::type_info)" (?registered_class_object@objects@python@boost@@YA?AV?$handle@U_typeobject@@@23@Utype_info@23@@Z)

I've read https://github.com/carla-simulator/carla/pull/2348 and https://github.com/carla-simulator/carla/issues/2339, but couldn't find the solution.
If you know how to solve this issue, please give me some advices.
Thanks in advance!

huobinzy, 2021-04-05:
Hello, Did you succeed?

corkyw10, 2021-04-06:
@EchoYJJ Thank you for the question. Are you able to post the full output of the `make PythonAPI` command. It sometimes finishes with a success message even if it was not successful. Can you also confirm if you are running the commands in the x64 Native Tools command prompt?

EchoYJJ, 2021-04-07:
> @EchoYJJ Thank you for the question. Are you able to post the full output of the `make PythonAPI` command. It sometimes finishes with a success message even if it was not successful. Can you also confirm if you are running the commands in the x64 Native Tools command prompt?

Hi! @corkyw10 Glad to receive your reply. I'm sure I run the commands in the x64 Native Tools command prompt.
And .egg file doesn't show up in PythonAPI\carla\dist.
The whole output is very long, there exists a lot of "warning C4819" : 该文件包含不能在当前代码页(936)中表示的字符。请将该文件保存为 Unicode 格式以防止数据丢失(This file contains characters that cannot be represented in the current code page (936).Please save this file as Unicode to prevent data loss)
Also, errors contain a lot of "error LNK2001": 无法解析的外部符号 "public: virtual unsigned int __cdecl boost::python::objects::py_function_impl_base::max_arity(void)const " (?max_arity@py_function_impl_base@objects@python@boost@@UEBAIXZ)

For sake of the big scale of the full output, I put it in the txt below. 
[full output.txt](https://github.com/carla-simulator/carla/files/6268123/full.output.txt)

By the way, I also try to run "make launch" anyway. It will open CarlaUE4, but nothing shows up.
[image]
I check my D:\carla\Unreal\CarlaUE4\Content\Carla, only a gz. file exists.
[image]

bernatx, 2021-04-07:
Hi,
It seems that the PowerShell command to extract the content .tar.gz has failed, because the file is so big. I suggest to install the 7zip.org application in the default folder and then the build system will use 7zip to extract the next time. In the mean time you can extract that file by hand.

About the unresolved external errors it seems you are using Python 3.8 instead the Python 3.7, or maybe you have more than one version. I suggest to run the **make** commands using the **ARGS="--python-version=3.7"** parameter to specify which version you want to use, otherwise it can be that the Boost library takes one version, and the CARLA build systems takes another version, and then we have mismatch.

Another thing to check is that you have installed the x64 version of Python and not the x32 version.

Regards

EchoYJJ, 2021-04-08:
Hi @bernatx  I do as you recommand, I download 7-Zip to "C:\Program Files (x86)\7-Zip". I don't know whether the position is right.
Then, I run **make PythonAPI ARGS="--python-version=3.7"** in VS2017 x64 tool.
However, the results seems to be no difference.

I also try to extract that file by hand, but it says "The File is corrupted". I notice the file is only 250KB, not so big like you say. Perhaps something went wrong with my tar file, do you know how to fix it?

Regards
[make PythonAPI output.txt](https://github.com/carla-simulator/carla/files/6275403/make.PythonAPI.output.txt)

goldenice123, 2021-04-12:
otherwise it can be that the Boost library takes one version, and the CARLA build systems takes another version, and then we have mismatch.

hi, i have the same problem, how to examine the version of boost version.

on@boost@@YA?AVobject@123@AEBV4123@AEBV?$handle@U_object@@@23@1@Z)
libcarla.obj : error LNK2001: 无法解析的外部符号 "class boost::python::api::object __cdecl boost::python::api::getattr(class boost::python::api::object const &,char const *)" (?getattr@api@python@boost@@YA?AVobject@123@AEBV4123@PEBD@Z)
libcarla.obj : error LNK2001: 无法解析的外部符号 "class boost::python::api::object __cdecl boost::python::api::operator<(class boost::python::api::object const &,class boost::python::api::object const &)" (??Mapi@python@boost@@YA?AVobject@012@AEBV3012@0@Z)
libcarla.obj : error LNK2001: 无法解析的外部符号 "void __cdecl boost::python::api::setitem(class boost::python::api::object const &,class boost::python::api::object const &,class boost::python::api::object const &)" (?setitem@api@python@boost@@YAXAEBVobject@123@00@Z)
libcarla.obj : error LNK2001: 无法解析的外部符号 "class boost::python::api::object __cdecl boost::python::api::operator==(class boost::python::api::object const &,class boost::python::api::object const &)" (??8api@python@boost@@YA?AVobject@012@AEBV3012@0@Z)
libcarla.obj : error LNK2001: 无法解析的外部符号 "public: __cdecl boost::python::api::object::object(class boost::python::handle<struct _object> const &)" (??0object@api@python@boost@@QEAA@AEBV?$handle@U_object@@@23@@Z)
libcarla.obj : error LNK2001: 无法解析的外部符号 "class boost::python::api::object __cdecl boost::python::api::getitem(class boost::python::api::object const &,class boost::python::api::object const &)" (?getitem@api@python@boost@@YA?AVobject@123@AEBV4123@0@Z)
build\lib.win-amd64-3.8\carla\libcarla.cp38-win_amd64.pyd : fatal error LNK1120: 77 个无法解析的外部命令
error: command 'D:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Community\\VC\\Tools\\MSVC\\14.16.27023\\bin\\HostX86\\x64\\link.exe' failed with exit status 1120

bernatx, 2021-04-12:
Hi, You can check which version of python goes to this file:
```carla\PythonAPI\carla\dependencies\lib\libboost_python37-vc141-mt-x64-1_72.lib```
In this case the python37 is for Python 3.7 version.
Maybe that helps

EchoYJJ, 2021-04-12:
> Hi, You can check which version of python goes to this file:
> `carla\PythonAPI\carla\dependencies\lib\libboost_python37-vc141-mt-x64-1_72.lib`
> In this case the python37 is for Python 3.7 version.
> Maybe that helps

Hi @bernatx , there do exist that file, could you please tell me how to check it.

7788boy, 2021-05-16:
Hi, 
I tried the following steps and it works.

1. Remove all the other versions of Python. (I uninstall Anaconda and reinstall Python 3.7.9 from Python.org.)
2. Add Python to path variable.
3. Delete the folders naming with "-install" in CARLA\Build
[image]
4. make clean
5. make PythonAPI

I hope it will help!

vibhoraggarwal, 2021-06-22:
>     1. Remove all the other versions of Python. (I uninstall Anaconda and reinstall Python 3.7.9 from Python.org.)
@7788boy after removing conda and running the `update.bat` file I received the following error. Did you also receive it?
```
Update.bat
20201222_232b876
http://carla-assets.s3.amazonaws.com/20201222_232b876.tar.gz
Downloading "http://carla-assets.s3.amazonaws.com/20201222_232b876.tar.gz"...
& : The term 'C:\Program Files\Anaconda3\Scripts\conda.exe' is not recognized as the name of a cmdlet, function,
script file, or operable program. Check the spelling of the name, or if a path was included, verify that the path is
correct and try again.
At C:\Users\AGV2FE\Documents\WindowsPowerShell\profile.ps1:4 char:4
+ (& "C:\Program Files\Anaconda3\Scripts\conda.exe" "shell.powershell"  ...
+    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (C:\Program File...ripts\conda.exe:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

7788boy, 2021-06-24:
> @7788boy after removing conda and running the `update.bat` file I received the following error. Did you also receive it?
>
> ```
> Update.bat
> 20201222_232b876 [...]

No, I didn't receive this.
Do you modify the path variable of Python?

vibhoraggarwal, 2021-07-20:
> No, I didn't receive this.
> Do you modify the path variable of Python?

Hi @7788boy, somehow I missed your reply. Currently, I have stopped using carla from source (using it from source is very inconsistent). I am just using the pre build version. If I build it again and face errors, I will post it here. 

P.S: I was using the source version in order to use the new vehicle import in carla

Yebihashu, 2021-09-01:
I succeeded to solve this problem (windows built version 9.12 ).
The problem was that I had more than one installed Python versions. 
Even I set windows PATH environment variable to point only on Python 3.8.10 which I use in this project, it seems that somehow the build process use also another Python version I had (Python 3.9.6) and a conflict was happen which cause to the link errors.

I do the follow steps:
(1)  Uninstall all other versions of python (in my case Python 3.9.6 )
(2) Took care that my environment variable PATH contains only the relevant folders needed for the Python I use (in my case 3.8.10 )
[image]
(3) Deleted the build folder of Carla - [Carla project folder]\Build
(4) Build Carla client again (make PythonAPI)
(5) Build Carle simulator server again (make launch)  

The boost used was indeed using python 3.8, you can see it according to the its file name as shown:
[image]

The dist folder was created and contains the correct version of .egg file as shown:
[image]

Note: At the beginning, I tried using python 3.9.6 , but I found that thing were failed here and there and therefor I moved to 3.8.10 which was ok.

nnorbert99, 2021-09-04:
I have the same exact problem and i tried to build it on an other computer and the same exact error message occured (so the problem is not with my PC ?), i only have one version of python (3.7) and VS2017. Is there any hope i can solve this ? (been stuck with this problem for 3 days now )

ZHAO-Garry, 2022-06-13:
Hi, I hit that issue and fix it by myself. I hope my experiences and fixes will be useful to everyone :)

### Condition

1. After hit **exit status 1120**, you should verifiy these files exists in path `{CARLA_ROOT}/PythonAPI/carla/`:
    - `xerces-c_3.lib`
    - `sqlite3.lib`
    - `proj.lib`
    - `libboost_*.lib`
 2. If you missing `libboot_*.lib` _(4 or 5 matches with *)_, it's caused by a mismatch between your version of Python and the version of Python used by Boost. In most cases it is because you are using **conda.** Please check **Fix for condition A** for help.
 3. If you **exactly missing** `xerces-c_3.lib`, `sqlite3.lib` and `proj.lib`, Please check **Fix for condition B** for help.

### Fix for Condition A

In this case, the paroblem raised by python environment. After my testing, **conda** and **pyenv** will have the same problem.  **For non-pro user**, you can use the checklist below:

1. Uninstall all the **Python** and **Python virtual environment manager** in your system. 
2. Remove all the **PATH** about Python in your system environment setting.
4. Directly **install Python** _(Availabe version: 3.7, 3.8, 3.9)_ with follow the official [documents](https://carla.readthedocs.io/en/latest/build_windows/).
5. Delete the entire **Build folder** `{CARLA_ROOT}/Build` in CARLA
6. Build now!

**For pro user**, you can check these things in your system:

1. **python version**: What's your python version and python path in your workspace?
2. **pip version**: Does the version of pip correspond to the version of python?
3. **py version**: Try command `py -V` to varify it. CARLA use `py` command to build. In **pyenv** cases, this will redirect to your system python setting.
4. **boost python**:  Boost will using the python version directly in system level. In **conda** and **pyenv** cases, this will use your system python setting. But you can also redirect it by adding setting `using python::"{PATH_TO_PYTHON}"` in `%homedrive%%homepath%\user-config.jam`.

**I did not succeed in successfully compiling the Python API using the Python virtual environment, so we can continue the discussion here :)**

### Fix for Condition B

In this case, the paroblem caused by a small 'quotation mark missing bug' in `{CARLA_ROOT}/Util/BuildTools/Setup.bat` _(line 195, 204, 213)_. This bug will let Windows failed to locate the path in some special cases, such like `"C:\Program Files (x86)"`. Using fixes below:

1. Open `{CARLA_ROOT}/Util/BuildTools/Setup.bat` and find line `195, 204, 213`
6. Adding quotation marks like below:
```
copy /Y "%INSTALLATION_DIR%\xerces-c-3.2.3-install\lib\xerces-c_3.lib" "%CARLA_PYTHON_DEPENDENCIES%\lib"
copy /Y "%INSTALLATION_DIR%\sqlite3-install\lib\sqlite3.lib" "%CARLA_PYTHON_DEPENDENCIES%\lib"
copy /Y "%INSTALLATION_DIR%\proj-install\lib\proj.lib" "%CARLA_PYTHON_DEPENDENCIES%\lib"
```
7. Build now!

KoheiAoyamaAB, 2023-09-13:
In my case, it works to run "make PythonAPI" two times without any changes. But I have no idea why it works.

TroyZhangFawang, 2024-08-21:
how to solve this problem? I face the same problem,救命

Jenson66, 2024-09-12:
> how to solve this problem? I face the same problem,救命

你解决了吗？我又碰到了一样的问题

closed 2026-11-13 by carla-triage[bot]