carla-simulator/carla #4062: Issue on Unreal Engine when build on linux

EchoYJJ, 2021-04-06:
Hi!I'm trying to build on linux ubuntu 18.04.
I follow the official instrction: https://carla.readthedocs.io/en/0.9.11/build_linux/

However, when I execute:
5. Open the Editor to check that UE has been properly installed.
cd ~/UnrealEngine_4.24/Engine/Binaries/Linux && ./UE4Editor
I meet some issues.
A new window show up, and if I try to open CarlaUE4,it shou the warning below:
"This project was created by different versions of Unreal Engine. Switching to this version will require recompiling your code project.
New features and improvements sometimes cause API changes, and you may need to modify the code before compilation. Content saved in a newer version of the editor will not be opened in the older version.
It is recommended that you open a copy of the project to avoid corruption of the original file."
[image]

And if I continue, the compile fails.
[image]

And if I try to run UE4Editor or CarlaUE4.uproject, it will turn out to be 
[image]

It seems to be version problem, but I don't know why it would happen and how to solve this.
If you know how to deal with it, please give me some advises, thanks!

corkyw10, 2021-04-06:
hi @EchoYJJ the line

`cd ~/UnrealEngine_4.24/Engine/Binaries/Linux && ./UE4Editor`

Is just to make sure that your compiled version of Unreal Engine will open without error, you have not built CARLA yet, only Unreal Engine. Once it Unreal opens, you can shut it down again and proceed to the next steps to actually build CARLA. Further down the tutorial you will see that the command to run CARLA in `make launch`

EchoYJJ, 2021-04-07:
@corkyw10 Hi! Glad to meet you (again), thanks a lot for your advises.
I do as you recommand. "make PythonAPI" receives a big success, the "carla-0.9.11-py3.6-linux-x86_64.egg" shows up in carla/PythonAPI/carla/dist.
However, when I keep running "make launch", error happens when building CarlaUE4 project: **fatal error: 'compiler/disable-ue4-macros.h' file not found**
The full output is listed below. Looking forward to your reply.
echoyjj@echoyjj-Lenovo-Legion-Y7000-2020:~/carla$ make launch
Setup.sh: llvm-8.0 already installed.
Setup.sh: boost-1.72.0-c8 already installed.
Setup.sh: rpclib-v2.2.1_c3-c8 already installed.
Setup.sh: gtest-1.8.1-c8 already installed.
Setup.sh: recast-cdce4e-c8 already installed.
Setup.sh: Libpng already installed.
Setup.sh: Xerces-c already installed.
Setup.sh: CARLA version 0.9.11-4-gfe3cb686-dirty.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Server.Release" configuration.
ninja: no work to do.
[0/1] Install the project...
-- Install configuration: "Server"
BuildLibCarla.sh: Success!
BuildCarlaUE4.sh: Using Unreal Engine at '/home/echoyjj/UnrealEngine_4.24'
BuildCarlaUE4.sh: Build CarlaUE4 project.
make[1]: 进入目录“/home/echoyjj/carla/Unreal/CarlaUE4”
bash "/home/echoyjj/UnrealEngine_4.24/Engine/Build/BatchFiles/Linux/RunMono.sh" "/home/echoyjj/UnrealEngine_4.24/Engine/Binaries/DotNET/UnrealBuildTool.exe" CarlaUE4Editor Linux Development  -project="/home/echoyjj/carla/Unreal/CarlaUE4/CarlaUE4.uproject" 

Running Mono...

Fixing inconsistent case in filenames.
Setting up Mono
~/UnrealEngine_4.24/Engine ~/carla/Unreal/CarlaUE4
Using 'git status' to determine working set for adaptive non-unity build (/home/echoyjj/UnrealEngine_4.24).
Building CarlaUE4Editor...
[Upgrade]
[Upgrade] Using backward-compatible build settings. The latest version of UE4 sets the following values by default, which may require code changes:
[Upgrade]     bLegacyPublicIncludePaths = false                 => Omits subfolders from public include paths to reduce compiler command line length. (Previously: true).
[Upgrade]     ShadowVariableWarningLevel = WarningLevel.Error   => Treats shadowed variable warnings as errors. (Previously: WarningLevel.Warning).
[Upgrade]     PCHUsage = PCHUsageMode.UseExplicitOrSharedPCHs   => Set in build.cs files to enables IWYU-style PCH model. See https://docs.unrealengine.com/en-US/Programming/BuildTools/UnrealBuildTool/IWYU/index.html. (Previously: PCHUsageMode.UseSharedPCHs).
[Upgrade] Suppress this message by setting 'DefaultBuildSettings = BuildSettingsVersion.V2;' in CarlaUE4Editor.Target.cs, and explicitly overriding settings that differ from the new defaults.
[Upgrade]
[Adaptive unity build] Excluded from PhysXVehicles unity file: PhysXVehicleManager.cpp
[Adaptive unity build] Excluded from VulkanRHI unity file: VulkanIndexBuffer.cpp, VulkanMemory.cpp, VulkanRHI.cpp, VulkanTexture.cpp, VulkanViewport.cpp
Performing 10 actions (8 in parallel)
[7/10] Compile Module.Carla.gen.5_of_6.cpp
[6/10] Compile Module.Carla.cpp
[5/10] Compile Module.Carla.gen.1_of_6.cpp
[2/10] Compile Module.Carla.gen.4_of_6.cpp
[3/10] Compile Module.Carla.gen.3_of_6.cpp
[4/10] Compile Module.Carla.gen.2_of_6.cpp
[1/10] Compile Module.Carla.gen.6_of_6.cpp
In file included from /home/echoyjj/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Carla/Module.Carla.cpp:2:
In file included from /home/echoyjj/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorBlueprintFunctionLibrary.cpp:8:
In file included from /home/echoyjj/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorBlueprintFunctionLibrary.h:9:
In file included from /home/echoyjj/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorDefinition.h:9:
/home/echoyjj/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorAttribute.h:9:10: fatal error: 'compiler/disable-ue4-macros.h' file not found
#include <compiler/disable-ue4-macros.h>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/echoyjj/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Carla/Module.Carla.gen.3_of_6.cpp:3:
In file included from /home/echoyjj/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Inc/Carla/FrictionTrigger.gen.cpp:8:
In file included from /home/echoyjj/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Trigger/FrictionTrigger.h:9:
In file included from /home/echoyjj/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/CarlaEpisode.h:9:
In file included from /home/echoyjj/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorDispatcher.h:9:
In file included from /home/echoyjj/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorDefinition.h:9:
/home/echoyjj/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorAttribute.h:9:10: fatal error: 'compiler/disable-ue4-macros.h' file not found
#include <compiler/disable-ue4-macros.h>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/echoyjj/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Carla/Module.Carla.gen.1_of_6.cpp:2:
In file included from /home/echoyjj/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Inc/Carla/ActorAttacher.gen.cpp:8:
/home/echoyjj/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Util/ActorAttacher.h:9:10: fatal error: 'compiler/disable-ue4-macros.h' file not found
[... cut ...]
/home/echoyjj/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorAttribute.h:9:10: fatal error: 'compiler/disable-ue4-macros.h' file not found
#include <compiler/disable-ue4-macros.h>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
1 error generated.
1 error generated.
1 error generated.
1 error generated.
1 error generated.
1 error generated.
Makefile:290: recipe for target 'CarlaUE4Editor' failed
make[1]: *** [CarlaUE4Editor] Error 5
make[1]: 离开目录“/home/echoyjj/carla/Unreal/CarlaUE4”
Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed
make: *** [launch] Error 2

corkyw10, 2021-04-07:
Hey @EchoYJJ glad that I could help. I am not sure now what the issue is, maybe @Axel1092 could help?

Axel1092, 2021-04-08:
Hi @EchoYJJ, the way you initially tried to open the Unreal editor probably polluted unreal's build system. Run `make clean` or `make hard-clean` and then `make launch`.
CARLA requires a lot of dependencies to be updated and properly linked with unreal, therefore, these commands make sure that the environment is ready. In the future you shouldn't try to open CARLA with `./UE4Editor` but rather through these commands.

EchoYJJ, 2021-04-09:
Hi @Axel1092, I follow your instruction"Run make hard-clean and then make launch."
Aftering receiving some warnings, the CaralaUE4 is opened.
Warning at 71%
[image]
Warning at 90%: Town 03.umap
[image]
And even if  there are many uasset under the Unreal/CarlaUE4/content/Carla.
The CarlaUE4 couldn't access them, and my CarlaUE4 is totally blank and couldn't import anything.
[image]

If you know how to solve this, please give me some advices. Many thanks!

Axel1092, 2021-04-13:
@EchoYJJ you are missing the assets. For this you need to run `./Update.sh` form the carla root. This will download them and place them in the unreal content folder.

EchoYJJ, 2021-04-14:
> @EchoYJJ you are missing the assets. For this you need to run `./Update.sh` form the carla root. This will download them and place them in the unreal content folder.

Hey! @Axel1092  I do as you recommend. I do run ./Update.sh before, but didn't put in the carla/Unreal/CarlaUE4/Content folder.(actually the offical guidance doesn't mention that, I think this procedure can be added.)

In this time, my CarlaUE4 can now access the uasset in the content folder and show them up in the content brower. However, when I try to open the basic Town0X.umap. The scene is totally black and white, I don't know what is going on there.
Please give me some advices, thanks in advance!

[image]
[image]

Axel1092, 2021-06-30:
Hi, maybe you need to wait for the shaders to finish compiling which takes some time for unreal to finish that.