carla-simulator/carla #4066: Issue on ScenarioRunner: vehicle doesn't move.

EchoYJJ, 2021-04-07:
Hi! I'm trying to use ScenarioRunner.
My setting:
CARLA 0.9.10 on Windows built from source
ScenarioRunner built from source

I've tried to run the example:  **python scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld --output**
However, only my ego vehicle show up, but doesn't move.
[image]

This is the output file.
[output.txt](https://github.com/carla-simulator/carla/files/6268717/output.txt)

Also, if I run the manual_control.py. The scene won't appear and command window keep saying "Waiting for the ego vehicle..."
[image]

If you know how to solve this issue, please give me some advices, thanks!

glopezdiest, 2021-04-08:
Hey @EchoYJJ, you just have to do both of those at the same time.

What the first command (`python scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld --output`) does is initialize the scenario. With the scenario ready, you can connect your agent and start moving the ego vehicle. In the example, we use the manual control (`python manual_control.py`), but you could use any other script that is capable of moving the ego vehicle.

What you are seeing from the first command output is the fact that scenarios have a timeout of around 1 minute, and after that time passes, they automatically end. What you have to do is run the scenario and then, the manual control on another terminal, before the timeout activates.

Hope this clarifies how to use ScenarioRunner.

EchoYJJ, 2021-04-08:
@glopezdiest Thanks a lot for your advices! I misunderstand these two commands as separate one.

I still remain one question about ScenarioRunner, my server FPS is very low(FPS=3). Do you know how to solve this?

EchoYJJ, 2021-04-11:
Hi @glopezdiest, I meet another issue, could you please give me some advices?
When running scenarios which require choreography of ego vehicle (i.e. in VehicleTurningRight, the route and speed of ego vehicle are arranged), the python manual_control.py couldn't result in expected situation. Even if I run python manual_control.py -a to set the autopilot mode. It won't act according to the python script.(like the video below, ego vehicle should turn right but it stops)
I wonder how to motivate the scenario according to the choreography in python script.

Looking forwards to your reply.
Regards

glopezdiest, 2021-04-12:
Hey @EchoYJJ, yes that is expected. As a general overview, the manual control is just a debugging agent that you can use to check how are the scenarios working, but it isn't a good agent anywhere else.

In this case, when you are setting the autopilot, the [TrafficManager](https://carla.readthedocs.io/en/latest/adv_traffic_manager/) is activated, which makes the vehicle randomly move through the town, respecting all traffic regulations. In other words, that's a feature of [CARLA's manual control](https://github.com/carla-simulator/carla/blob/master/PythonAPI/examples/manual_control.py) which just doesn't have anything to do with the scenario itself (to be honest, we should remove that functionality of the manual control at SR, as it is just confusing)

EchoYJJ, 2021-04-12:
> move that functionality

Hi @glopezdiest . So if not manual_control.py, how can I get my ego vehicle to the right place at the right time(in the video case, it should be the position of cyslist). I think the expected route of ego vehicle is arranged in the python script, how can I motivate it?

glopezdiest, 2021-04-12:
Well, you have to create your own agent that does that. First of all, you can get the route from any script using [`CarlaDataProvider.get_ego_vehicle_route()`](https://github.com/carla-simulator/scenario_runner/blob/master/srunner/scenariomanager/carla_data_provider.py#L395). 

Then, you should create an agent that uses that route to move. [Here](https://github.com/carla-simulator/carla/blob/master/PythonAPI/examples/automatic_control.py) is an example script that uses different agents to move around town, and by modifying their destination function (around [line 700](https://github.com/carla-simulator/carla/blob/master/PythonAPI/examples/automatic_control.py#L700)), you should be able to move through the specified route.

EchoYJJ, 2021-04-12:
Hi @glopezdiest thanks a lot for your advices. For my scenario, ego vehicle have no difference from other vehicles, I don't want to control the ego vehicle. So I want to define the behaviors of ego vehicle in python script and the motivation should be merely generate a camera on any vehicle to provide a vision for me.
Whether do you know a simpler way to achieve this?
Regards

EchoYJJ, 2021-04-12:
I'm a little confused. In this example: https://www.youtube.com/watch?v=Mmb3qz7G-W0
1.If the vehicle in the screen is ego vehicle, how can it set to be automatic accroding to the script?
2.If the vehicle in the screen is other vehicle and the car passing by is ego vehicle, how can the camera set on other vehicle?
[image]

glopezdiest, 2021-04-12:
> For my scenario, ego vehicle have no difference from other vehicles, I don't want to control the ego vehicle

Then you can use the [WaypointFollower](https://github.com/carla-simulator/scenario_runner/blob/master/srunner/scenariomanager/scenarioatomics/atomic_behaviors.py#L1809) to control the ego vehicle through a given route (which will probably need to calculate).

> If the vehicle in the screen is ego vehicle, how can it set to be automatic accroding to the script, or even ego vehicle is controlled by hand?

The ego is controlled by hand

> If the vehicle in the screen is other vehicle and the car passing by is ego vehicle, how can the camera set on other vehicle?

The agent connects to the ego using [this loop](https://github.com/carla-simulator/scenario_runner/blob/master/manual_control.py#L99). You'd have to do something similar, but with another vehicle. The rolename can be set when spawning the vehicle (when calling [this function](https://github.com/carla-simulator/scenario_runner/blob/master/manual_control.py#L99))

EchoYJJ, 2021-04-15:
Hi @glopezdiest , thanks for your advices, changing the agent to another vehicle seems to be a little hard for me.
Only thing I want to do is make the vehicle shows up in the screen is defined by the python script. So, I needn't any control of ego vehicle, I just focus on the interaction between other vehicles.
So, as an alternative, in order to change the spectator from ego vehicle to other vehicle, I add following code in class WorldSR(World) of manual_control.py. 

_other_vehicle = self.world.get_actors().filter('vehicle.nissan.patrol')
spectator = self.world.get_spectator()
transform = other_vehicle[0].get_transform()
spectator.set_transform(carla.Transform(transform.location + carla.Location(z=50),
                                       carla.Rotation(pitch=-90)))_

However, the vehicle shows up in screen is still ego vehicle. Actually, I can't find where manual_control.py define the spectator.
Could you please give me some advices on how to correctly adjust the spectator?

glopezdiest, 2021-04-15:
The spectator doesn't play any role here. What the manual control does is set an RGB camera at the connected vehicle, so you aren't seeing the spectator, but the sensor itself. 

> However, the vehicle shows up in screen is still ego vehicle

Same as before, if you haven't changed the connected vehicle to the `vehicle.nissan.patrol`, the camera will still be created at the ego, so you'll still see the ego vehicle

glopezdiest, 2021-11-24:
Closing this as it seems solved. Feel free to reopen it if needed

closed 2021-11-24 by glopezdiest