carla-simulator/carla #4067: Lidar and semantic lidar results are different at one simulation

Kaiwind88, 2021-04-07:
Hi,

I found an issue that when I run a simple scenario with a regular lidar and a semantic lidar and save the outputs at the same time,. But the point clouds from the two lidars are not the same. I am expecting identical results. Does anyone have this issue as well and how to solve it?

Thanks.
Sample results are below: 
Left is from semantic lidar and right is from regular lidar.

[image]

The codes are attached below.

`import glob
import os
import sys
try:
    sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    pass
import carla
import argparse
import random
import time
import numpy as np
import cv2

IM_WIDTH = 640
IM_HEIGHT = 480

parent_path = "../Outputs/"

# Todo
# add argument input

class Lidar:
    def __init__(self):
        self.channels = 32
        self.max_range = 100.0
        self.atmosphere_attenuation_rate = 0.0
        self.upper_fov = []
        self.lower_fov = []
        self.points_per_second = 21700
        self.rotation_freq = []
        self.noise_std = []
        self.dropoff_general_rate = []
        self.dropoff_intensity_limit=[]
        self.dropoff_zero_intensity=[]

def process_img(image):
    i = np.array(image.raw_data)
    i2 = i.reshape((IM_HEIGHT, IM_WIDTH, 4))
    i3 = i2[:, :, :3]
    cv2.imshow("", i3)
    cv2.waitKey(1)
    return i3/255.0

def process_lidar(data):
    
    # data.save_to_disk(parent_path + '%06d.ply' % data.frame_number)

    frame_number = data.frame_number

    outFilePathTemp  = parent_path + f'lidar-{frame_number}_temp.carlapcd'
    outFilePath = parent_path + f'lidar-{frame_number}.pcd'
    out = np.frombuffer(data.raw_data, dtype=np.dtype('f4'))
    out = np.reshape(out, (int(out.shape[0] / 4), 4))
    number_points = out.shape[0]
    
    np.savetxt(outFilePathTemp, out, fmt='%1.8f')

    # Convert to PCD format
    list_of_lines=["# .PCD v0.7 - Point Cloud Data file format","VERSION 0.7","FIELDS x y z intensity",
     "SIZE 4 4 4 4", "TYPE F F F F","COUNT 1 1 1 1","WIDTH "+str(number_points),"HEIGHT 1","VIEWPOINT 0 0 0 1 0 0 0","POINTS " +str(number_points),"DATA ascii"]

    with open(outFilePathTemp, "r") as read_obj, open(outFilePath, "w") as write_obj:
        for line in list_of_lines:
            write_obj.write(line+'\n')
        for line in read_obj:
            write_obj.write(line)
    os.remove(outFilePathTemp)
    # print("Conver to pcd successfully")

def process_lidar_sematic(data):
    
    # data.save_to_disk(parent_path + 'semantic-%06d.ply' % data.frame_number)
    frame_number = data.frame_number
    # print(frame_number)
    outFilePathTemp  = parent_path + f'lidar-semantic-{frame_number}_temp.carlapcd'
    outFilePath = parent_path + f'lidar-semantic-{frame_number}.pcd'
    
    dt = np.dtype([
        ('x', np.float32), ('y', np.float32), ('z', np.float32),
        ('CosAngle', np.float32), ('ObjIdx', np.uint32), ('ObjTag', np.uint32)])
    out = np.frombuffer(data.raw_data, dtype=dt)
    
    points = np.array([out['x'], out['y'], out['z'], out['ObjTag']]).T
    # print(out_pcd)
    number_points = points.shape[0]
    
    np.savetxt(outFilePathTemp, points, fmt="%1.8f "*3 + " %d")
    
    # Convert to PCD format
    list_of_lines=["# .PCD v0.7 - Point Cloud Data file format","VERSION 0.7","FIELDS x y z intensity",
     "SIZE 4 4 4 4", "TYPE F F F F","COUNT 1 1 1 1","WIDTH "+str(number_points),"HEIGHT 1","VIEWPOINT 0 0 0 1 0 0 0","POINTS " +str(number_points),"DATA ascii"]

    with open(outFilePathTemp, "r") as read_obj, open(outFilePath, "w") as write_obj:
        for line in list_of_lines:
            write_obj.write(line+'\n')
        for line in read_obj:
            write_obj.write(line)
    os.remove(outFilePathTemp)

def RemoveFiles():
    print("Remove old files")
    files = glob.glob(parent_path+"*")
    for f in files:
        os.remove(f)

def main(args):
    delta = 1.0/args.fps
    actor_list = []
    try:
        if args.delete_file:            
            RemoveFiles()
        print("Start !")
        client = carla.Client('localhost', 2000)
        client.set_timeout(2.0)

        world = client.get_world()
        settings = world.get_settings()
        
        settings.fixed_delta_seconds = delta
        world.apply_settings(settings)
       

        blueprint_library = world.get_blueprint_library()
        vehicle_bp = blueprint_library.filter('model3')[0]

        spawn_point_list = world.get_map().get_spawn_points()
        spawn_point = spawn_point_list[0]
        vehicle = world.spawn_actor(vehicle_bp, spawn_point)
        # vehicle.apply_control(carla.VehicleControl(throttle=1.0, steer=0.0))
        vehicle.set_autopilot(True)  # if you just wanted some NPCs to drive.

        actor_list.append(vehicle)

        #################### add lidar #########################

        lidar = Lidar()
        lidar.rotation_freq = args.fps
        lidar_bp = blueprint_library.find('sensor.lidar.ray_cast')
        lidar_bp.set_attribute('channels', f'{lidar.channels}')
        lidar_bp.set_attribute('range', f'{lidar.max_range}')
        lidar_bp.set_attribute('rotation_frequency', f'{lidar.rotation_freq}')
        lidar_bp.set_attribute('points_per_second', f'{lidar.points_per_second}')
        lidar_spawn_point = carla.Transform(carla.Location(x=-0.5, z=1.8))
        lidar = world.spawn_actor(lidar_bp, lidar_spawn_point, attach_to=vehicle)
        actor_list.append(lidar)
        # lidar.listen(lambda data: data.save_to_disk('_out/%06d.ply' % data.frame_number))
        

        lidar_gt = Lidar()
        lidar_gt.rotation_freq = args.fps
        lidar_gt_bp = blueprint_library.find('sensor.lidar.ray_cast_semantic')
        lidar_gt_bp.set_attribute('channels', f'{lidar_gt.channels}')
        lidar_gt_bp.set_attribute('range', f'{lidar_gt.max_range}')
        lidar_gt_bp.set_attribute('rotation_frequency', f'{lidar_gt.rotation_freq}')
        lidar_gt_bp.set_attribute('points_per_second', f'{lidar_gt.points_per_second}')
        lidar_gt_spawn_point = carla.Transform(carla.Location(x=-0.5, z=1.8))
        lidar_gt = world.spawn_actor(lidar_gt_bp, lidar_gt_spawn_point, attach_to=vehicle)
        actor_list.append(lidar_gt)
        # lidar_gt.listen(lambda data: data.save_to_disk('_out/%06d.ply' % data.frame_number))
        lidar.listen(lambda data: process_lidar(data))
        lidar_gt.listen(lambda data: process_lidar_sematic(data))

    #################### add RGB camera #########################
        # https://carla.readthedocs.io/en/latest/cameras_and_sensors
        # get the blueprint for this sensor
        # blueprint = blueprint_library.find('sensor.camera.rgb')
        # # change the dimensions of the image
        # blueprint.set_attribute('image_size_x', f'{IM_WIDTH}')
        # blueprint.set_attribute('image_size_y', f'{IM_HEIGHT}')
        # blueprint.set_attribute('fov', '110')

        # # Adjust sensor relative to vehicle
        # spawn_point = carla.Transform(carla.Location(x=2.5, z=0.7))

        # # spawn the sensor and attach to vehicle.
        # sensor = world.spawn_actor(blueprint, spawn_point, attach_to=vehicle)

        # # add sensor to list of actors
        # actor_list.append(sensor)

        # # do something with this sensor
        # sensor.listen(lambda data: process_img(data))

        time.sleep(1)

    finally:
        print('destroying actors')
        for actor in actor_list:
            actor.destroy()
        print('done.')

if __name__ == "__main__":
    argparser = argparse.ArgumentParser(
        description=__doc__)
    argparser.add_argument('-d',
        '--delete_file',
        default=False,
        type=bool,
        help='Delete files in folder: '+ parent_path)

    argparser.add_argument('-f',
        '--fps',
        default= 10,
        type=float,
        help='FPS')
    args = argparser.parse_args()
    try:
        main(args)
    except KeyboardInterrupt:
        print(' - Exited by user.')
`

DSantosO, 2021-04-26:
Hello @Kaiwind88,

The LiDAR sensor incorporates some point dropoff to simulate the loss of points in actual sensors. If you want to really compare both sensors, you can tune the parameters to avoid dropping any points. Check the details [here](https://carla.readthedocs.io/en/latest/ref_sensors/#lidar-sensor). 
I have checked the case with a LiDAR with parameters: `dropoff_general_rate=0.0` and `dropoff_intensity_limit=0.0` and the point count match. If you see it differently, please let us know.

closed 2021-09-03 by DSantosO

seungtae-Han, 2023-04-25:
> Hello @Kaiwind88,
>
> The LiDAR sensor incorporates some point dropoff to simulate the loss of points in actual sensors. If you want to really compare both sensors, you can tune the parameters to avoid [...]

Hello, I am writing a comment because I have a question.
If I mount a lidar sensor and a semantic lidar sensor in the same position and set the lidar sensor to dropoff_general_rate=0.0 and dropoff_intensity_limit=0.0, will the coordinates of the points between the two sensors be output identically?

For example:
LiDAR sensor: [[x1, y1, z1, i1], [x2, y2, z2, i2], ...]
Semantic LiDAR sensor: [[x1, y1, z1, label], [x2, y2, z2, label], ...]

The reason I am asking this question is that I want to build a dataset using the semantic lidar sensor, but there is a problem that the output of the semantic lidar sensor does not include intensity. If the coordinates are output identically as in Example 1, I plan to include the intensity of the lidar sensor output in the semantic lidar sensor output.

Example 2) Integrated output: [[x1, y1, z1, i1, label], [x2, y2, z2, i2, label], ...]

Please reply,,,, thank you!!

Paolo08, 2023-11-13:
Hello @seungtae-Han, any updates? Also, did you include the _intensity_ value into the semantic lidar? Thanks 😄

seungtae-Han, 2023-11-13:
> Hello @seungtae-Han, any updates? Also, did you include the _intensity_ value into the semantic lidar? Thanks 😄

Hi, @Paolo08 .
I tested it, and when I set dropoff_general_rate=0.0 and dropoff_intensity_limit=0.0, the number and coordinates of points in LIDAR and SEMANTIC LIDAR were generated identically. 
Also, when I output the point cloud of Semantic LIDAR as a pcd file, I could include the intensity values of LIDAR.

Paolo08, 2023-11-13:
@seungtae-Han I see, thanks for replying. I'm interested to add _intensity_ to semantic lidar implementation. Did you do that perhaps? If so, can you share it?