carla-simulator/carla #4070: carla0.9.11 + carsim running problems

whuxhzhang, 2021-04-07:
Ubuntu18.04
carla-0.9.11

I use carsim in carla-0.9.11：
self.player.enable_carsim("/home/zhangxinhong/CarSim_2020.0_Ubuntu_x86_64_128709_with_vsvisualizer/CarSim_2020.0/Example_Data/carla_sim.sim")
But I can't get the effect in the document：
[image]

In addition, I want to know what method or interface is used by carla and carsim to transfer data described in the document：
[image]

FENGShuanglang, 2021-09-23:
你好，请问你是怎么在ubuntu中安装carsim的，是买的正版吗？

panpan-77, 2022-03-04:
> 你好，请问你是怎么在ubuntu中安装carsim的，是买的正版吗？

Hi, I have the same problem. Have you solved it？

whuxhzhang, 2022-03-04:
您好，

我是申请的试用版，申请成功会有操作文档，可以根据文档安装。

263868161, 2022-05-19:
请问如何申请试用版？

closed 2023-06-02 by stale[bot]