carla-simulator/carla #4084: Get bone transforms

mrcoee, 2021-04-12:
I want to spawn pedestrians in a natural walking position without them moving. I read that one can use the 'apply_control' function to manipulate the skeleton of the pedestrians. To do this, I need the bone transforms of the corresponding walking animations. Is there a way to obtain them using some carla function?

I figured out that by using the UE editor i can display the bone transform in an arbirtrary frame of the animation. However, so far i could not find a way to save them in a file which i can open with python and read the transforms from. Unfortunately, I do not have any experience with the UE editor.

Schuck84, 2021-05-03:
Hi,
It would be great to get that function.
Could you give some hint to implement it self?
What files must be modified?
How is the information transferred from server to client?
What UE C++ function must be called?

hvelesaca, 2021-05-17:
I am interested to obtain the list of bone transforms and subsequently get the joints of a specific walker. Is it possible?

closed 2023-06-02 by stale[bot]