carla-simulator/carla #4085: mismatch between cpp and python

guygo12345, 2021-04-13:
Hey!

Running on carla docker 0.9.11.
When running the following code -
...
```
traffic_lights = world.get_actors().filter('traffic.traffic_light')
            for tl in traffic_lights:
                tl.set_state(carla.TrafficLightState.Green)
```
...
I'm getting this error - 
`tl.set_state(carla.libcarla.TrafficLightState.Green)
Boost.Python.ArgumentError: Python argument types in
    TrafficLight.set_state(TrafficLight, TrafficLightState)
did not match C++ signature:
    set_state(carla::client::TrafficLight {lvalue}, carla::rpc::TrafficLightState state)
`

Why would that happen? It seems like I'm using the function in the right way...

Thanks for helping!

guygo12345, 2021-04-13:
OK, I manged to get over the bug. It was something  about the way I imported carla.

closed 2021-04-13 by guygo12345

JasonBourne1998, 2023-02-03:
Hi @guygo12345 , could you please specify how to resolve this ?