carla-simulator/carla #4087: Quickstart not working

arihant2math, 2021-04-13:
The documentation in the quickstart does not specify how to install the package carla, so the examples in pythonapi/examples exit with an error. However spectator mode still works.

Windows 10, Carla 0.9.10

LucasFKobernic, 2021-04-16:
Hi arihant2math, 

Using Carla Package u just need to unzip the file you have downloaded. 

Then you need to start CARLA (CarlaUE4.exe) first. And then you can run any of the examples inside the PythonAPI / examples folder.

arihant2math, 2021-04-19:
I did that, the examples still do not work.

corkyw10, 2021-04-20:
Hi @arihant2math, what error are you getting?

arihant2math, 2021-04-20:
I'm getting this error when I run `spawn_npc.py`: `ModuleNotFoundError: No module named 'carla'`

corkyw10, 2021-04-21:
Can you make sure that the PYTHONPATH is set correctly by running the following, or adding it to your `.bashrc` file:

```
export PYTHONPATH=$PYTHONPATH:"$HOME/<path/to/carla>/PythonAPI/carla/dist/<name_of_egg_file>.egg"
export PYTHONPATH=$PYTHONPATH:"$HOME/<path/to/carla>/carla"
export PYTHONPATH=$PYTHONPATH:"$HOME/<path/to/carla>/carla/PythonAPI/"
```

arihant2math, 2021-04-24:
@corkyw10 I am using windows.

corkyw10, 2021-04-30:
Hi @arihant2math, sorry for the misunderstanding. Can you look in the directory `carla\PythonAPI\carla\dist\` and check the `.egg` file there? It should look like this `carla-<version>-py3.7-win-amd64.egg`. The part `py3.7` will indicate the version of Python that should be used, it's normally 3.7 in the package versions. Can you confirm that you are running the scripts with Python 3.7?

arihant2math, 2021-04-30:
I was using python 3.9, using 3.7 fixed the problem.

closed 2021-04-30 by arihant2math