carla-simulator/carla #4089: make launch error /home/loc/carla/Util/BuildTools/BuildCarlaUE4.sh: line 161:  3527 Killed                  ${GDB} ${UE4_ROOT}/Engine/Binaries/Linux/UE4Editor "${PWD}/CarlaUE4.uproject" ${RHI} Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed make: *** [launch] Error 137

HankYuan1101, 2021-04-14:
Hi, there
I follow the Linux build instruction and I got an error while executing make launch.
The error shown as follow:

```
/home/loc/carla/Util/BuildTools/BuildCarlaUE4.sh: line 161:  3527 Killed                  ${GDB} ${UE4_ROOT}/Engine/Binaries/Linux/UE4Editor "${PWD}/CarlaUE4.uproject" ${RHI}
Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed
make: *** [launch] Error 137
```

OS: ubuntu 18.04
carla version: 0.9.11
unreal version: 4.24

any suggestion is welcome, thanks!

onebitme, 2021-04-15:
`gcc -v
g++ -v`

Can you please check the version of gcc, 
I am currently using v 7.5.0

HankYuan1101, 2021-04-19:
> `gcc -v g++ -v`
> 
> Can you please check the version of gcc,
> I am currently using v 7.5.0

My gcc version is also 7.5.0
gcc --version                                
gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
PURPOSE.
g++ --version                                
g++ (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0

Are those difference with your version?

corkyw10, 2021-04-20:
Hi @Hakhyun-Kim could you attach the full output of any error you get after running

`make rebuild`

HankYuan1101, 2021-04-23:
Hi @corkyw10, thanks for your feedback.

There is a lot of "can't find file" in the output. I'm not sure the connection between missing files and this error message. 
But I export the UnrealUngine directory in the bashrc. Is there other directory setting?

Here is my full output after running your suggestion

```
BuildLibCarla.sh: Success!
BuildCarlaUE4.sh: Using Unreal Engine at '/home/loc/UnrealEngine_4.24'
BuildCarlaUE4.sh: Generate Unreal project files.
Attempting to set up UE4 pretty printers for gdb (existing UE4Printers.py, if any, will be overwritten)...
	updated UE4Printers.py
	found necessary entries in ~/.gdbinit file, not changing it.

Setting up Unreal Engine 4 project files...

Fixing inconsistent case in filenames.
Setting up Mono
Discovering modules, targets and source code for project...
While compiling /home/loc/carla/Unreal/CarlaUE4/Intermediate/Build/BuildRules/CarlaUE4ModuleRules.dll:
/home/loc/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.Build.cs(39,9) : warning CS0618: `UnrealBuildTool.ModuleRules.Definitions' is obsolete: `The 'Definitions' property has been deprecated. Please use 'PublicDefinitions' instead.'
Generating data for project indexing... 98%----------------------------------------------
CarSim OFF
100%
Discovering modules, targets and source code for project...
Generating data for project indexing... 98%----------------------------------------------
CarSim OFF
100%
Writing project files... 100%
Discovering modules, targets and source code for project...
Generating data for project indexing... 98%----------------------------------------------
CarSim OFF
100%
Discovering modules, targets and source code for project...
Generating data for project indexing... 98%----------------------------------------------
CarSim OFF
100%
Discovering modules, targets and source code for project...
Generating data for project indexing... 98%----------------------------------------------
CarSim OFF
100%
Discovering modules, targets and source code for project...
Generating data for project indexing... 98%----------------------------------------------
CarSim OFF
100%
Writing project files... 100%
BuildCarlaUE4.sh: Build CarlaUE4 project.
make[1]: Entering directory '/home/loc/carla/Unreal/CarlaUE4'
bash "/home/loc/UnrealEngine_4.24/Engine/Build/BatchFiles/Linux/RunMono.sh" "/home/loc/UnrealEngine_4.24/Engine/Binaries/DotNET/UnrealBuildTool.exe" CarlaUE4Editor Linux Development  -project="/home/loc/carla/Unreal/CarlaUE4/CarlaUE4.uproject" 

Running Mono...

Fixing inconsistent case in filenames.
Setting up Mono
~/UnrealEngine_4.24/Engine ~/carla/Unreal/CarlaUE4
Using 'git status' to determine working set for adaptive non-unity build (/home/loc/UnrealEngine_4.24).
Creating makefile for CarlaUE4Editor (no existing makefile)
----------------------------------------------
CarSim OFF
Creating makefile for UnrealHeaderTool (no existing makefile)
------- Build details --------
Using toolchain located at '/home/loc/UnrealEngine_4.24/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v15_clang-8.0.1-centos7/x86_64-unknown-linux-gnu'.
Using clang (/home/loc/UnrealEngine_4.24/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v15_clang-8.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang++) version '8.0.1' (string), 8 (major), 0 (minor), 1 (patch)
Using bundled libc++ standard C++ library.
Using lld linker
Using llvm-ar : /home/loc/UnrealEngine_4.24/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v15_clang-8.0.1-centos7/x86_64-unknown-linux-gnu/bin/llvm-ar
Using fast way to relink  circularly dependent libraries (no FixDeps).
------------------------------
Parsing headers for CarlaUE4Editor
[... 1588 lines cut ...]
[2021.04.23-06.04.46:739][  0]LogLoad: (Engine Initialization) Total time: 95.96 seconds
[2021.04.23-06.04.46:739][  0]LogLoad: (Engine Initialization) Total Blueprint compile time: 0.00 seconds
[2021.04.23-06.04.46:765][  0]LogSlate: Took 0.000369 seconds to synchronously load lazily loaded font '../../../Engine/Content/Slate/Fonts/Roboto-Regular.ttf' (155K)
[2021.04.23-06.04.46:769][  0]LogSlate: Took 0.001211 seconds to synchronously load lazily loaded font '../../../Engine/Content/Slate/Fonts/Roboto-Bold.ttf' (160K)
[2021.04.23-06.04.46:776][  0]LogSlate: Took 0.000720 seconds to synchronously load lazily loaded font '../../../Engine/Content/Editor/Slate/Fonts/FontAwesome.ttf' (139K)
[2021.04.23-06.04.47:233][  0]LogMaterial: Display: Missing cached shader map for material LevelGridMaterial2, compiling. 
[2021.04.23-06.04.47:359][  0]LogRenderer: Reallocating scene render targets to support 1168x584 Format 10 NumSamples 1 (Frame:1).
[2021.04.23-06.04.47:454][  0]LogStaticMesh: Allocated 1006x425x128 distance field atlas = 104.4Mb, with 86 objects containing 58.7Mb backing data
[2021.04.23-06.04.47:658][  0]LogSlate: Took 0.000076 seconds to synchronously load lazily loaded font '../../../Engine/Content/Slate/Fonts/Roboto-Regular.ttf' (155K)
[2021.04.23-06.04.48:182][  0]LogLinux: Scale at X=0.000000, Y=0.000000: 1.000000 (monitor=#0, HDPI=96.000000 (horz scale: 1.000000), VDPI=96.000000 (vert scale: 1.000000))
[2021.04.23-06.04.48:183][  0]LogLinux: Scale at X=960.000000, Y=540.000000: 1.000000 (monitor=#0, HDPI=96.000000 (horz scale: 1.000000), VDPI=96.000000 (vert scale: 1.000000))
[2021.04.23-06.04.48:430][  0]LogContentStreaming: Texture pool size now 4000 MB
[... cut ...]
[2021.04.23-06.04.50:367][ 31]LogLinux: Scale at X=0.000000, Y=0.000000: 1.000000 (monitor=#0, HDPI=96.000000 (horz scale: 1.000000), VDPI=96.000000 (vert scale: 1.000000))
[2021.04.23-06.04.50:367][ 31]LogLinux: Scale at X=960.000000, Y=540.000000: 1.000000 (monitor=#0, HDPI=96.000000 (horz scale: 1.000000), VDPI=96.000000 (vert scale: 1.000000))
/home/loc/carla/Util/BuildTools/BuildCarlaUE4.sh: line 161: 27885 Killed                  ${GDB} ${UE4_ROOT}/Engine/Binaries/Linux/UE4Editor "${PWD}/CarlaUE4.uproject" ${RHI}
Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed
make: *** [launch] Error 137

```

corkyw10, 2021-04-23:
When you ran the command from this part of the [doc](https://carla.readthedocs.io/en/0.9.11/build_linux/#unreal-engine), did Unreal start up successfully? 

```
cd ~/UnrealEngine_4.24/Engine/Binaries/Linux && ./UE4Editor
```

And you downloaded all the assets correctly and they were extracted to `carla/Unreal/CarlaUE4/Content`?

seuwcs, 2021-06-17:
> When you ran the command from this part of the [doc](https://carla.readthedocs.io/en/0.9.11/build_linux/#unreal-engine), did Unreal start up successfully?
>
> ```
> cd [...]

Hello, I have met a question that after run the code:
cd ~/UnrealEngine_4.24/Engine/Binaries/Linux && ./UE4Editor
in the terminal. The Unreal Engine could not start up, and the output is as below.

[image]

[image]

Could you give me some suggestions on solving this question?  Thank you !

Ryan-Doren, 2021-07-01:
Hello @HankYuan1101, I ran into the same problem as yours, have you solved it yet?

closed 2023-06-02 by stale[bot]