carla-simulator/carla #4090: Modifying existing Cars

jhermen-List, 2021-04-14:
Hi There, thanks a lot for the great work you are doing on Carla.

I found the documentation on how to create and add new vehicles, but is there a way to modify existing vehicles, like adding a lamp, replacing texture or similar smaller changes without redesigning a completely new car? 

Thanks a lot 
       Johannes

onebitme, 2021-04-15:
Hi there,

For modifying mesh (e.g. you want to add a 3d model of a sensor to an already existing vehicle in Carla), simply export the vehicle in .fbx, import the file in blender, add 3d mesh (be sure that normals of the vertices are towards out) and then follow the guide.
For textures, google "how to change textures in UE4" will be sufficient for you :)
For more, simply check for the solutions which are not carla specific but for UE4.

Also, I might help if you can describe what you are trying to achieve in more detail :)

jhermen-List, 2021-04-15:
Thanks a lot, that sound do-able! I wil ltry that.
The idea is to add a kind of signaling/lights to the front of the Car that can be triggered via the API.

i guess the VehicleLightState.Special vars can somehow be used for that.

gsmario, 2021-04-20:
> Thanks a lot, that sound do-able! I wil ltry that.
> The idea is to add a kind of signaling/lights to the front of the Car that can be triggered via the API.
> 
> i guess the VehicleLightState.Special vars can somehow be used for that.

yeah you can add as many lights as you want, but remember to set also de lights in unreal engine. Rightnow our lights are controlled via API as well so you probably won't have any problem.

jhermen-List, 2021-07-21:
> The idea is to add a kind of signaling/lights to the front of the Car that can be triggered via the API.
> i guess the VehicleLightState.Special vars can somehow be used for that.

So, i archived to modify the vehicle blueprint and add some meshes for the places where i want to show something. 
Now the question is how to exchange the material on this mesh via the VehicleLightState.Special vars to show one or another image on the front. Any hints for that?

[image]

Thanks a lot
       Johannes

closed 2021-11-17 by germanros1987