carla-simulator/carla #4092: How to give a path to my vehicle and pedestrian?

LucasFKobernic, 2021-04-14:
Hi everyone,

I'm new by CARLA and have some questions about controlling walkers and vehicles.

I want to create some specific acident scenarios involving pedestrians. 

In the simplest case i need the pedestrian to follow a specific path and the car another one. How can i give a specific path to my car and to my pedestrian? (For the pedestrian i tried to use WalkerAIController and the method go_to_location, but by reaching the first goal  my pedestrian becomes automacally another one. For the car i need to use scenario runner right?). 

In the most complex case i want to dynamically change the next goal point, based on information from the environment (like the position of the ego vehicle). Is that possible? Can i also attacht a camera to my pedestrian?

Thanks for the help,

Regards, 

Lucas

---
OS: Windows 10
Carla Package 0.9.11
scenario runner 0.9.11
Python 3.7

glopezdiest, 2021-04-15:
> For the car i need to use scenario runner right

Yes, I'd recommend using ScenarioRunner as, while it might take a bit to understand, there are lots of simple behavior that will help you create the accident traffic scenario you are aiming for. 

> For the pedestrian i tried to use WalkerAIController and the method go_to_location, but by reaching the first goal my pedestrian becomes automacally another one

No, you can't use the `WalkerAIController`, as it is meant to be used to populate the simulation with pedestrians. That controller will force walkers to move through the sidewalks, sometimes crossing the road, but you won't be able to make a pedestrian move through a specific route through he road. 

> In the most complex case i want to dynamically change the next goal point, based on information from the environment (like the position of the ego vehicle). Is that possible?

You theoretically can do that, yeah. I'm not sure how you'd do it, as I'm not fully seeing your use-case, but you can definitely do it

> Can i also attacht a camera to my pedestrian?

Again, yeah. you can either create a behavior inside ScenarioRunner that does that or, you can also do something similar [ScenarioRunner's manual control ](https://github.com/carla-simulator/scenario_runner/blob/master/manual_control.py), where you sync a camera to a specific vehicle ([this loop here](https://github.com/carla-simulator/scenario_runner/blob/master/manual_control.py)

LucasFKobernic, 2021-04-15:
Thanks for the infos :) They were very helpfull

closed 2021-04-15 by LucasFKobernic