carla-simulator/carla #4095: Segmentation fault (core dumped) while running CarlaUE4.sh file.

bijayaniSMishra, 2021-04-15:
Hi Team, 

I am trying to install and deploy Carla in the AWS EC2 instance. I am able to install Carla 9.10 successfully. but while running the simulation using the CarlaUE4.sh file I get the below error. 

ubuntu@ip-172-31-25-195:~$ cd /opt/Carla-simulator/bin
ubuntu@ip-172-31-25-195:/opt/carla-simulator/bin$ SDL_VIDEODRIVER=offscreen ./CarlaUE4.sh
4.22.3-0+++UE4+Release-4.22 517 0
Disabling core dumps.
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119
Malloc Size=115872 LargeMemoryPoolOffset=247008
Failed to find symbol file, expected location:
"/opt/carla-simulator/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping.sym"
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119
Malloc Size=115872 LargeMemoryPoolOffset=247008
Failed to find symbol file, expected location:
"/opt/carla-simulator/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping.sym"
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119
Malloc Size=103376 LargeMemoryPoolOffset=234512
Failed to find symbol file, expected location:
"/opt/carla-simulator/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping.sym"
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped) 

This is the first time I am using Carla. Can you please help in solving this issue. I am using the below GitHub link for installing and Deploying Carla in AWS EC2. 

https://github.com/jbnunn/CARLADesktop 

Also Please let me know if you have any documentation for installing and deploying Carla on any Cloud platform.

Thanks,
Bijayani

QianwenZhu, 2021-06-17:
the same problem as mine.

If I execute this command:
"SDL_VIDEODRIVER=offscreen SDL_HINT_CUDA_DEVICE=0 ./CarlaUE4.sh --world-port=2000 -opengl"

the message is :
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119
Malloc Size=74976 LargeMemoryPoolOffset=206112
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.

Else if  I execute this command:
" ./CarlaUE4.sh -opengl"

the message is as following and it exits immediately:
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.

abdooo786, 2022-04-22:
> 

i'm facing the same problem, did you find any solution ?

closed 2026-11-13 by carla-triage[bot]