carla-simulator/carla #4100: Scenario runner and ros bridge single scenarios compatibility

mortenfc, 2021-04-18:
Hi!

Scenario Runner's single and grouped `.xml` scenarios specified with `--scenario` flag don't work with Ros Bridge.
Please fix this.

Best regards,
Morten

glopezdiest, 2021-04-20:
Hey @mortenfc, could you please give more information about your exact test as well as the errors you are getting? I'm also adding @joel-mb

mortenfc, 2021-04-22:
Ok,

`--waitForEgo` flag in Scenario Runner does make it not spawn Ego correctly, but after Ros Bridge spawns ego and it starts moving as can be seen in RViz, the leading vehicle is never spawned and the scenario execution never happens.
Scenario Runner does not work without `--reloadWorld` for me. I don't understand why there is an option to run without this flag.
After making Ros Bridge attach with its launch parameter and Scenario Runner do the spawning, then it works.
IF Scenario Runner was started BEFORE Ros Bridge.
And :group scenarios don't work. (Respawning ego with scenario-runner does not reset sensors etc. in Ros Bridge).
I made a workaround for this by publishing a reset cmd from Scenario Runner on scenario reset, which resets all sensors
in Ros Bridge python `spawn_objects_XXX.py`. 
But this only works when the world object stays the same (when the map does not change).
And it also messes with the ros clock -> it goes backwards :( -> because scenario runner reloads the world, because `--reloadWorld` flag is required.

closed 2023-06-02 by stale[bot]