carla-simulator/carla #4101: No rendering mode in carla 11

mortenfc, 2021-04-18:
No rendering mode does not work in carla 11.
Camera sensor is still ticking away after applying `settings.no_rendering_mode = True` to the world.
Tried in multiple different ways with example scripts in PythonAPI.

lgalardini, 2021-04-18:
Hi there, 

I'm using Carla .11 as well, under Windows.

I used the spawn_npc example to develop my data collection script, and I'm able to disable the rending with the following snippets (it is in the definition of the **CarlaSyncMode** context manager):
`self.frame = self.world.apply_settings(carla.WorldSettings(
            no_rendering_mode=**True**,
            synchronous_mode=True,
            fixed_delta_seconds=self.delta_seconds))`

As workaround, have you tried to disable the rendering using the python config script in the util folder? This works for me, just run it after the client is started: `config.py --no-rendering`

mortenfc, 2021-04-18:
Thanks, I'll try that.
Does it require synchronous mode now?
I tried `config.py --no-rendering`, it does not work for me.
That's perhaps because it does not apply sync mode as well.
https://github.com/carla-simulator/carla/blob/master/PythonAPI/examples/no_rendering_mode.py#L944

I think the documentation and python example scripts should be updated to reflect that synchronous mode is required for it to work. 
https://carla.readthedocs.io/en/0.9.11/adv_rendering_options/#no-rendering-mode

lgalardini, 2021-04-18:
> Thanks, I'll try that.
> Does it require synchronous mode now?
> I tried `config.py --no-rendering`, it does not work for me.
> That's perhaps because it does not apply sync mode as well.
> https://github.com/carla-simulator/carla/blob/master/PythonAPI/examples/no_rendering_mode.py#L944
> 
> I think the documentation and python example scripts should be updated to reflect that synchronous mode is required for it to work, if it is intended to only work for sync mode.
> https://carla.readthedocs.io/en/0.9.11/adv_rendering_options/#no-rendering-mode

Personally speaking I don't think sync mode is mandatory, I was just underlining the piece of code in **spawn_npc.py** example that allows you to switch between no-rending modes. But I've not that much experience to bet on this.
I don't know where you apply the settings, maybe your settings are overwritten somehow. Try with the workaround I suggested, if it work I think it is matter just of placing the code in the correct place.

mortenfc, 2021-06-10:
I tried that line you provided with also enabling sync mode. It does not work.
I tried searching in ROS bridge if it overwrites the settings somewhere.
A recursive code search for `apply_settings` yielded no fruit.

closed 2023-06-02 by stale[bot]