carla-simulator/carla #4103: Error about "make launch" in building CARLA

by-seong-me, 2021-04-19:
I'm trying to build CARLA with Unreal Engine
When I ran "make launch", I got these errors

**error: misleading indentation; statement is not part of the previous 'if' [-Werror,-Wmisleading-indentation]

error: no member named 'GameDir' in 'FPaths'

fatal error: 'PhysicsEngine/BodySetupEnums.h' file not found**

I just followed the instructions in https://carla.readthedocs.io/en/latest/build_linux/
But, why such errors happened?
Could somebody help me?

corkyw10, 2021-04-20:
Hello @by-seong-me, be aware that the **latest** version of the documentation details how to install the version of CARLA that is still in development. If you want to install the most recent release of CARLA, follow the docs for that version (0.9.11) by selecting it in the bottom right corner of the screen on the documentation page.

Can you confirm which version of Unreal Engine you are using? Bear in mind that 0.9.11 uses UE 4.24 and the development version is using UE 4.26.

by-seong-me, 2021-04-21:
@corkyw10 Thank you! I changed version of UE as 4.24 and the same error is no longer displayed. But I encountered with another issues like

/home/msc/Downloads/carla/Util/BuildTools/BuildCarlaUE4.sh: line 161:  5569 Killed                  ${GDB} ${UE4_ROOT}/Engine/Binaries/Linux/UE4Editor "${PWD}/CarlaUE4.uproject" ${RHI}
Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed
make: *** [launch] Error 137

So I'm trying to re-install CARLA 0.9.11. 
I just hope the installation to be success someday... 

Anyway thank you @corkyw10 !!

closed 2021-04-21 by by-seong-me

GeraldGuo7, 2021-04-29:
I finally met the same problem and confused about these two fatal errors. 
Personally,  this icon should be remarkable or place in the top of the Carla website with the red color.

by-seong-me, 2021-05-02:
@GeraldGuo7 Yes, I totally agree with that.
And also, I figured out above issues.
I recommend that "do not use virtual environment when building Carla"

GeraldGuo7, 2021-05-08:
Right, and I didn't delve into the error files so I have no idea about what happened.
I think the latest one is still under development so some paras and files are missing.
I decided to use the version 0.9.11 and everything looks well until now.

GeraldGuo7, 2021-05-08:
> @GeraldGuo7 Yes, I totally agree with that.
> And also, I figured out above issues.
> I recommend that "do not use virtual environment when building Carla"
Right, and I didn't delve into the error files so I have no idea about what happened.
I think the latest one is still under development so some paras and files are missing.
I decided to use the version 0.9.11 and everything looks well until now.

GeraldGuo7, 2021-05-08:
@by-seong-me Right, and I didn't delve into the error files so I have no idea about what happened.
I think the latest one is still under development so some paras and files are missing.
I decided to use the version 0.9.11 and everything looks well until now.

ShoofLLC, 2021-06-12:
This issue should be reopened, the file [PrepareAssetsForCookingCommandlet.cpp](https://github.com/carla-simulator/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Commandlet/PrepareAssetsForCookingCommandlet.cpp) still uses the old `GameDir` method instead of `ProjectDir`, even though according to the dev team, they're using 4.26, so that's definitely an issue.

See also [this forum post](https://forums.unrealengine.com/t/4-25-replacement-for-gamedir/143527/2)

bernatx, 2021-06-14:
Hi,
I think there is some confusion here. In current last development status, we are using ProjectContentDir() on PrepareAssetsForCookingCommandlet.cpp file (for more than 1 year), but we still use **/Game/** which is not deprecated. Only some functions from the FPaths object has been deprecated, but those has been already fixed.

ShoofLLC, 2021-06-14:
@bernatx oh, sorry for the confusion, it was because the "dev" branch is actually the real current branch and not "master"

libing0811, 2021-07-02:
I come up with same issue now, after checking the instruction of windows build. I find the  0.9.11 and latest instruction both need us to git master branch of carla. So I think maybe current master version is also can only support UE4.24. Maybe the dev branch can use UE4.26, not master?
I think someone need to modify the instruction in latest version.

corkyw10, 2021-07-07:
@libing0811 you are right that the master branch currently uses UE4.24 and the dev branch uses 4.26. There is a warning at the top of the latest documentation that explains it references the "dev" branch. This documentation will become master on the next release. Do you have any suggestions on how to make this more clear in the latest documentation other than the top page warning?

libing0811, 2021-07-08:
> @libing0811 you are right that the master branch currently uses UE4.24 and the dev branch uses 4.26. There is a warning at the top of the latest documentation that explains it references the "dev" [...]

I think maybe in the latest version builing page, we can modify the _**git clone https://github.com/carla-simulator/carla_** add the branch of dev, so the user can done load the dev branch of carla.
Thank you.

hexrefugee, 2022-07-16:
> @corkyw10 Thank you! I changed version of UE as 4.24 and the same error is no longer displayed. But I encountered with another issues like [...]

hi, I met the same error. Are your fix this question?
The first time I try "make launch". I can open the UE4's interface.But after a few mins, the terminal show errors just like you quote.And the next time I can't open the UE4' interface it just show errors.
Can you give me some advice to fix it?

taotaohuang001, 2022-10-21:
I am having the same issue. Did you have any updates?

.../BuildCarlaUE4.sh: line 172: 114275 Killed                  ${GDB} ${UE4_ROOT}/Engine/Binaries/Linux/UE4Editor "${PWD}/CarlaUE4.uproject" ${RHI} make: *** [Util/BuildTools/Linux.mk:7: launch] Error 137

> hi, I met the same error. Are your fix this question? The first time I try "make launch". I can open the UE4's interface.But after a few mins, the terminal show errors just like you quote.And the [...]

hexrefugee, 2022-10-21:
I fixed. the ue's edit should compare with carla edit.

| |
***@***.***
|
|
***@***.***
|

---- Replied Message ----
| From | ***@***.***> |
| Date | 10/21/2022 13:14 |
| To | ***@***.***> |
| Cc | ***@***.******@***.***> |
| Subject | Re: [carla-simulator/carla] Error about "make launch" in building CARLA (#4103) |

I am having the same issue. Did you have any updates?

.../BuildCarlaUE4.sh: line 172: 114275 Killed ${GDB} ${UE4_ROOT}/Engine/Binaries/Linux/UE4Editor "${PWD}/CarlaUE4.uproject" ${RHI} make: *** [Util/BuildTools/Linux.mk:7: launch] Error 137

@corkyw10 Thank you! I changed version of UE as 4.24 and the same error is no longer displayed. But I encountered with another issues like
/home/msc/Downloads/carla/Util/BuildTools/BuildCarlaUE4.sh: line 161: 5569 Killed GDB{UE4_ROOT}/Engine/Binaries/Linux/UE4Editor "PWD/CarlaUE4.uproject"{RHI} Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed make: *** [launch] Error 137
So I'm trying to re-install CARLA 0.9.11. I just hope the installation to be success someday...
Anyway thank you @corkyw10 !!

hi, I met the same error. Are your fix this question? The first time I try "make launch". I can open the UE4's interface.But after a few mins, the terminal show errors just like you quote.And the next time I can't open the UE4' interface it just show errors. Can you give me some advice to fix it?

TeamF111, 2023-06-14:
> @libing0811 you are right that the master branch currently uses UE4.24 and the dev branch uses 4.26. There is a warning at the top of the latest documentation that explains it references the "dev" [...]

hi, my system is Ubuntu 20.04, i installed UE4.26 and CARLA 0.9.14. I have the error: 
/home/uos/carla/Util/BuildTools/BuildCarlaUE4.sh: line 185: 63757 Killed                  ${GDB} ${UE4_ROOT}/Engine/Binaries/Linux/UE4Editor "${PWD}/CarlaUE4.uproject" ${RHI} ${EDITOR_FLAGS}
make: *** [Util/BuildTools/Linux.mk:7: launch] Error 137

I think the reason is due to the dev or master branch is that right?
can i use UE4.24 and CARLA 0.9.11 on Ubuntu 20.04 which is master branch?

thank you

zehranrgi, 2023-06-17:
> 

what do you mean?

zehranrgi, 2023-06-17:
> I fixed. the ue's edit should compare with carla edit. | | ***@***.*** | | ***@***.*** |
> […](#)

what do you mean?

romantic-black, 2024-05-17:
> hi, my system is Ubuntu 20.04, i installed UE4.26 and CARLA 0.9.14. I have the error: /home/uos/carla/Util/BuildTools/BuildCarlaUE4.sh: line 185: 63757 Killed ${GDB} [...]

The issue was caused by insufficient memory. After increasing the swap space, the problem was resolved.

minhaj6, 2024-11-25:
Can confirm the insufficient memory issue. I got this error with 16GB of memory. The error went away when 32GB of RAM was used instead.