carla-simulator/carla #4107: Retrieve image data from the spectator's view?

Stagakis, 2021-04-20:
Hello team,

I am working on an AR visualization project where I want to show information to the driver's windshield about obstacles, cars, pedestrians, etc.

Since I want the testing phase of my project to have, ideally, real-time performance, I was wondering whether I could get the image data of the spectator view from the client API. This way I could at least skip one RGB image sensor and since RGB image rendering is very resource-heavy on Carla, this can come in handy.

The off-screen option in Carla docs -> Rendering options specifies that the OpenGL API (which has worse performance) must be used and indeed it crashes when I try to run it with Vulkan. Since it seems that I cannot get rid of the spectator, I thought of using it instead of a camera.

In the docs, the spectator is just an Actor class, not a Sensor, so as far as I know there is no listen() function.

So the question is, does a workaround exist? If not, do you know how I can run the simulation in real-time with an RGB image sensor? I get just over 60 fps with manual_control.py + spawn_npc.py on a pc with Ryzan 3900x and Nvidia 3090.

DSantosO, 2021-04-26:
Hello @Stagakis,
No, I don't think you can recover the data for the spectator. As far as I see, I can recommend two options.
 - Take the spectator to a place where the rendering is trivial. For example, below the ground where no objects can be seen. In this case, this should provide a speed-up and the other camera should effectively be like your only camera.
 - In the current `dev` branch, we have uploaded to the 4.26 version of Unreal. This adds support for the off-screen with Vulkan so you can get rid of the spectator and use Vulkan for rendering. You can try this in our `dev` branch or in the [overnight build](http://carla-releases.s3.amazonaws.com/Linux/Dev/CARLA_Latest.tar.gz).

Stagakis, 2021-04-26:
Thank you very much, I didn't think to try the overnight build.

closed 2021-04-26 by DSantosO