carla-simulator/carla #4108: "DebugHelper.draw_point" function only draw black points in 0.9.11.

Chiakiaki, 2021-04-20:
Hi,
I'm doing trajectory planning using Carla 0.9.11. The "draw_point" function can only plot with black color and I don't know why. I previously use 0.9.8, and there is no such issue.

E.g. the following two lines of code all draw black points. 
world.debug.draw_point(w, size = 0.1,color = carla.Color(0,0,0,0), life_time = xxx)
world.debug.draw_point(w, size = 0.1,color = carla.Color(255,0,0,255), life_time = xxx)

Anyone idea about this?

glopezdiest, 2021-05-05:
Hey @Chiakiaki, indeed we saw it after the release of 0.9.11 and we fixed it later on (current *dev* doesn't have this issue). However, I can't seem to find the commit where we fixed it, and we swapped to unreal 4.26 so it might be tricky to just upgrade to current *dev*.

Chiakiaki, 2021-05-05:
@glopezdiest Thank you for your reply. You said you wrapped it to unreal 4.26. Could I simply replace the 'CarlaUE4' folder with current dev, without changing other parts such as 'pythonAPI' folder?

glopezdiest, 2021-05-05:
Not sure, but I really don't recommend it

Chiakiaki, 2021-09-24:
@glopezdiest Hi, is this pull request "https://github.com/carla-simulator/carla/pull/3342"  related to this issue? Or do you know how to fix this issue if I build from source? Thank you.

PS: I recently tried version 0.9.10.1 but such issue also exist (0.9.10.1 is used in the Carla Challenge).

glopezdiest, 2021-09-27:
Not sure. I don't think that PR is the one that fixes it, as it was done before the release of 0.9.10.1, so technically, it should be done and working, which isn't the case. As I previously said here, it was at some point fixed but I still don't know when that happened. Maybe you can manually change that value and see if that helps?

closed 2023-06-02 by stale[bot]