carla-simulator/carla #4110: how to spawn multiple spectator

nuomizai, 2021-04-20:
I spawned multiple ego vehicles in one carla world through one carla client. And I wanted to check out every vehicles' state at the same time. But I found that the function CarlaDataProvider.get_world().get_spectator() only returned one spectator. Does anyone konw how can I use multiple spectators at the same time? Or I should spawn multiple cameras above each vehicles?

glopezdiest, 2021-04-27:
Hey @nuomizai, as far as I know, the spectator is a camera which is the camera Unreal uses to show the scene, and there's only one. You can manually move and rotate it to your heart's content, which is definitely not something you can do with any other sensor.

If you want to show multiple ego vehicles, you are gonna have to create multiple cameras and render them using something like pygame (like the manual control does)

nuomizai, 2021-04-27:
> Hey @nuomizai, as far as I know, the spectator is a camera which is the camera Unreal uses to show the scene, and there's only one. You can manually move and rotate it to your heart's content, which [...]

Thanks for your reply, @glopezdiest ! I've created multiple cameras as you suggested and use opencv-python to render them.

closed 2021-04-27 by glopezdiest

SExpert12, 2023-12-13:
> I spawned multiple ego vehicles in one carla world through one carla client. And I wanted to check out every vehicles' state at the same time. But I found that the function [...]

How did you spawn multiple vehicles? Can you please share your code. I want to spawn multiple vehicles with few distance apart and I am trying but not working.