carla-simulator/carla #4124: Make launch error :  "PhysicsEngine/BodySetupEnums.h" not found

aakashsehgal, 2021-04-21:
Building Carla 0.9.11 on windows 10.
Unreal Engine 4.26
Python 3.7.9

Building the unreal engine succeeds.

While building the Carla server with make launch i receive an error  

> "C:/Tools/Carla/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/OpenDrive/OpenDriveGenerator.cpp(19): fatal error C1083: Cannot open include file: 'PhysicsEngine/BodySetupEnums.h': No such file or directory
> make: *** [CarlaUE4Editor] Error 6"
> 

I also checked in the` "C:\UnrealEngine\Engine\Source\Runtime\Engine\Private\PhysicsEngine"` but no such file `"BodySetupEnums.h"` is present. 

How can i get this file ?

Axel1092, 2021-04-22:
Hi @aakashsehgal, you are building a version of carla that works with UE 4.24 and not 4.26, the errors you see are the result of the incompatibility. You need to switch to UE4.24.

jbenzhhn, 2021-04-22:
I ran into the same problem because the https://carla.readthedocs.io/en/latest/ is made up for the next carla versino 0.9.12 (which is not released yet). Make sure to select the fitting carla version down below. 
[image]

aakashsehgal, 2021-04-23:
ok...so i need to shift to UE4.24 and Carla 0.9.11 right ?

jbenzhhn, 2021-04-23:
Exactly :) For me it is working right now

closed 2021-05-05 by corkyw10

NikhilKalghatgi, 2021-05-13:
@jbenzhhn @aakashsehgal how do we shift to UE4.24? I am having a hard time doing that, please help here

jbenzhhn, 2021-05-13:
Hi @NikhilKalghatgi ;) Remove the whole carla folder from your installation of latest build and just try to follow the  [carla wiki build instructions](https://carla.readthedocs.io/en/0.9.11/build_linux/) with the corresponding version. In the moment its carla 0.9.11.

NikhilKalghatgi, 2021-05-14:
> Hi @NikhilKalghatgi ;) Remove the whole carla folder from your installation of latest build and just try to follow the [carla wiki build instructions](https://carla.readthedocs.io/en/0.9.11/build_linux/) with the corresponding version. In the moment its carla 0.9.11.

Thanks it worked :)

ucbmpc2021, 2021-11-08:
Hi! @Axel1092  @jbenzhhn  I am building Carla 0.9.9 and using UE4.24. However I still got the exact same error. Any clue why this might happened? Much thanks!