carla-simulator/carla #4125: Pedestrian/Walker stopping at RED traffic signal

cyber-loop, 2021-04-21:
Hi all,

I want to know if there is way to control the pedestrian walking on the junction i.e. if traffic signal is red then pedestrian should stop crossing the road and when traffic signal is green then continue walking. Can someone point me out if there is any function for that in Python? Thanks

CL

glopezdiest, 2021-04-27:
Hey @cyber-loop. There is not :slightly_frowning_face:, you'd have to implement it. If you are using the *WalkerAIController*, pedestrians will ignore traffic lights. You might be able to manage it using their *WalkerControl*, but it's not a trivial task.

glopezdiest, 2021-11-24:
Seems solved, so closing the issue.

closed 2021-11-24 by glopezdiest

bittuh, 2022-04-29:
> Seems solved, so closing the issue.

Hi Carla team,

I'm still looking for the way to enable a smarter walker control (e.g. stopping at red light) with carla 0.9.12,
could you provide more info about how this is solved?