carla-simulator/carla #4126: Custom materials with xodr

NielsPichon, 2021-04-22:
Hi!
I am trying to build on top of the osm/xodr map loader to be able to train some AI agent with openScenarios which use the said xodr maps. However, my goal being to be able to navigate using vision as the primary source of information, I would need to improve on the graphics side of things.
After a bit of digging, it seems like when loading a xodr file with `generate_opendrive_world`, one of 2 materials is assigned, either "road" or "crosswalk". This will be translated accordingly in the RecastBuilder I think.

But I am not sure where to go from here... Where are these materials defined? How could I add my own materials (possibly procedural), to add lane lines for instance or house textures on blocks I would spawn on the side of the road?

Thanks for the help,
Niels

NielsPichon, 2021-04-22:
In the mean time I found this [AActor](https://github.com/carla-simulator/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/OpenDrive/OpenDriveGenerator.cpp). Am I right in assuming this is contained in the OpenDriveMap.umap and that this will generate a procedural mesh for the roads with no texture and default vertex color? 

This would mean that I could possibly add my textures in there for it to be loaded at runtime?

algababr, 2021-07-01:
Hey Niels sorry for the late response, but we have been a bit busy lately preparing for the new release. Are you still looking into it?

NielsPichon, 2021-07-01:
Hi Roel,
Yes this would still be of interest to me to be able to add custom materials to xodr maps. I haven't found the solution on my own just yet. Any clue?

algababr, 2021-07-01:
When we use "make import" for importing maps into carla (along with the .xodr information), we specify the materials that are going to be applied through our [commandlet](https://github.com/carla-simulator/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Commandlet/PrepareAssetsForCookingCommandlet.cpp) which derives from Unreal's commandlet. In the constructor we define those materials for the roads, the markings, sidewalks...That's how we modify anything imported into carla.

If you are using "generate_opendrive_world" separately from this process, I'll have to look into it. Let me know if this solution works for you :)

algababr, 2021-07-06:
Hey Niels! Did you have the chance to test this out?

NielsPichon, 2021-07-08:
Hi!
Yes it seems to have done the trick. Thanks for the help.

closed 2021-07-08 by NielsPichon

NielsPichon, 2022-10-11:
> Hi @NielsPichon<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FNielsPichon&data=05%7C01%7C%7C2707ee05d439400fc1bb08da6352049f%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637931501155784422%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=XkXbpAh2iYai9jX5cggtit46SlG9O0KW%2FK9RH3feW7o%3D&reserved=0>, I'm looking into procedural level generation from xodr, too. What did you end up doing? Just make import, provide the .xodr and leave out the .fbx?

Hi @brscholz. I am afraid I kind of gave up on this as I could not figure out a simple way to do it. Good luck