carla-simulator/carla #4132: Changing Towns in CARLA 0.9.10

NishanthARao, 2021-04-23:
Hi all.

I just wanted to know, how do you change towns during the execution of command ```make launch```? While launching the binary version, we can change towns via ```./CarlaUE4.sh Town07```, but that doesn't work in this case i.e., ```make launch Town07```. I wasn't able to find it in any of the documentations.

I am using CARLA 0.9.10 (Ubuntu 18.04) with Unreal Engine 4.24.3

Thanks a lot!

onebitme, 2021-04-26:
[image]

I am not sure if this solves your problem but whenever I want to change the towns in UE4 Editor, I manually change the level from content browser

jayabrata97, 2021-04-30:
Hi,
 I am using CARLA 0.9.10. I have used quick start installation on Windows10.  Can anyone please tell me how can I use Town06 in Windows? I can use Town04 by running python3 config.py --map Town04, but cannot use the same command for Town06.

closed 2021-08-05 by corkyw10