carla-simulator/carla #4136: Logging object list of all dynamic objects

TKeutgens, 2021-04-23:
I'm trying to run an OpenScenario file using the scenario runner and want to record an object list of all the dynamic objects with (position, velocity, acceleration, heading) to use as reference data. 

I know that there is the Carla recorded (https://carla.readthedocs.io/en/0.9.11/adv_recorder/), but using the show_recorder_file_info.py I am only able to retrieve position information about the objects and not the other attributes. Is there any other information about the usage of the csv output of the client.show_recorder_file_info. 

As a lot of people in automated driving development work with object lists it would be great to have an easy option to log and extract data from the simulation for further analysis.

glopezdiest, 2021-05-12:
Hey @TKeutgens, if you start the recorder with the `additional_data` set, all information about the actors will be recorded (without it, only the minimum information necessary to replay it will be saved):
`client.start_recorder(filename, additional_data=True)`

You can check [SR's metrics parser](https://github.com/carla-simulator/scenario_runner/blob/master/srunner/metrics/tools/metrics_parser.py#L282), along with the [metrics log](https://github.com/carla-simulator/scenario_runner/blob/master/srunner/metrics/tools/metrics_log.py#L30), to have an idea on how to parse the *show_recorder_file_info.py* **into a dictionary**.

Apart from that, there really is only the `client.show_recorder_file_info()` as a means to parse the *.log* file, and that only allows its output to be a *txt* file, with the specific format I'm sure you already saw.

glopezdiest, 2021-11-24:
Closing this issue as it seems to be solved. Feel free to reopen it if need be.

closed 2021-11-24 by glopezdiest